portapack-mayhem/mayhem-firmware #1358: scanner app long description overwrites the text of MCLR button

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
23/08/06
----

**Describe the bug**
A clear and concise description of what the bug is. 

long description on a line (size > 25 cararacters )overwrites the MCLR field text

**To Reproduce**
Steps to reproduce the behavior:

file containing this line f=1000,d=abcdefghijklmnopqrstuvwxyz1

**Expected behavior**
A clear and concise description of what you expected to happen.

We should print only the max descriptor  size

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-08-10:
Use the Recon app.
A too long description will not overwrite GUI.

NotherNgineer, 2023-08-14:
Resolved by PR #1370

closed 2023-08-14 by NotherNgineer