portapack-mayhem/mayhem-firmware #1362: scanner app file with one range, and single frequ, the range is not scanned

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
23/08/06
----

**Describe the bug**
 

a file containing one line as a range, and other lines with single freq, the range is not scanned

**To Reproduce**
Steps to reproduce the behavior:

use this file:

f=90000,d=90000
f=91000,d=91000
a=100000000,b=108000000,d=NEVER
f=92000,d=92000

Max = 3, should be more 

**Expected behavior**

Should scan all the lines

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2023-08-07:
This is by design. You have to hit the "Search" button to search the Range. Changing to Enhancement request.

NotherNgineer, 2023-08-07:
Seems like a duplicate of #1361

jLynx, 2023-08-07:
Closing due to being a duplicate

closed 2023-08-07 by jLynx