portapack-mayhem/mayhem-firmware #1363: freqman why "category" is printed on top of screen, shoud be F: like file, to leave more space for the file name

alain00091, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

as day one we use "category" to show the name of the selected file

**Describe alternatives you've considered**

Change "category" to "F:", it will leave more space to show full name of files ...
example with SDCARD  USA-PAGER-UPPER(.TXT)

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

jLynx, 2023-08-07:
We could also look adding some sort of marquee scrolling which would also address the problem

gullradriel, 2024-03-11:
will be solved by https://github.com/portapack-mayhem/mayhem-firmware/pull/1977