portapack-mayhem/mayhem-firmware #1373: No configuration holding

samukas81, 2023-08-13:
The same problem of not holding the configuration came back.  I'm using the latest night version.

NotherNgineer, 2023-08-14:
Presuming that you're referring to "same problem" as #1369.  Could you provide more details?

samukas81, 2023-08-14:
I did the procedure that Brumi suggested, but it only worked for a period, I turned off the equipment and about 30 minutes later I went to turn it on and I had the same problem.  It simply does not maintain configuration!

jLynx, 2023-08-14:
@samukas81 check the voltage of your RTC battery, I am assuming its flat if its not holding the config

samukas81, 2023-08-14:
She is normal, loading without problems.

jLynx, 2023-08-14:
But above you just said it was not holding the config, so then this issue is solved?

samukas81, 2023-08-14:
You mean the battery behind the display?

jLynx, 2023-08-14:
the small round battery. If that is flat, then you will lose your config

samukas81, 2023-08-14:
Ok, I will check and post the result.

samukas81, 2023-08-14:
It's the battery, it's only 500mV, I'll replace it and see if it goes back to normal.  Thank you for now.

jLynx, 2023-08-14:
Yeah that will be the reason why. I'll close this issue for now. Please let us know if that doesnt solve it

closed 2023-08-14 by jLynx

samukas81, 2023-08-14:
Problem solved, it was the battery.