portapack-mayhem/mayhem-firmware #1375: Inconsistent M4 CPU utilization in proc_capture

NotherNgineer, 2023-08-15:
Running Audio RX app in WFM mode, it has been noted that the M4 CPU utlization varies between 4% and 100% when simple changes are made to the proc_capture module.

Comparing some recent builds which had minor changes to proc_capture:
- n_230814:  M4 CPU ~4-10%
- n_230807:  M4 CPU ~55-100%  (user @pedroCX486 also reported WFM Audio stuttering after PR #1352 in this nightly)
- v1.7.4:  M4 CPU 100%

Additionally, after exiting the Audio app and running it again or temporarily switching the modulation types and then going back to WFM, M4 CPU utilization changes substantially.

Is the M4 CPU utilization value completely inaccurate?  Or are the numbers real and perhaps this is related to the CPU cache?  Perhaps proc_capture size is growing and unable to fit in CPU cache any more...

CC:  @Brumi-2021

NotherNgineer, 2023-08-16:
The low M4 CPU utilization % values (<=10%) in n_230814 & n_230816 were due to an issue in the DFU overlay screen trying to display 6 digits in a 5-wide text field (this issue was made visible by #1370).

The DFU overlay is resolved by #1383, **BUT still there is the question regarding why M4 utilization varies between 50%-100%.**

gullradriel, 2023-09-09:
In the nighlty at the date of this post, on audio app in WFM mode:
-M0 is 100% 
-M4 is varying around 60% but not that much

Brumi-2021, 2023-09-10:
Hi @gullradriel , I tested today's nightly 23-09-10th , and I got same results. 
From first boot , launching directly  Audio App , with last use WFM , 
I got  M4  usage 100%,  (but possible to recover changing WFM ==>NFM ==>WFM again  (it drops around 56 to 60%) 

(If last use demod. mode  was NFM , then launching Audio App ,goes to NFM then ==>WFM (and ok, <60%) .

So in my H1R1, I think , we still have the same problem