portapack-mayhem/mayhem-firmware #1377: freqman should support comment line

alain00091, 2023-08-16:
**Describe the solution you'd like**
Comment Line starting with "#" should be supported because you have more space/characters to write informations
"#" comment line are used in some files in the SDCARD that you provide

today it is in grey on the first screen ( why not)  and if you go to EDIT mode it is RAW in the entry type and error,
it should be COMMENT TYPE and valid

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

NotherNgineer, 2023-08-16:
Note that # is expected for comment lines, such as in the file USA-PAGER-* files in the standard SD card image.  Freqman uses entry type "RAW" for comments, by design.

However, I agree with you that it should not say "Error".  I also see the the length is being truncated to 17 characters when edited, which could also be improved.

alain00091, 2023-08-17:
Yes I checked the code of freqman and the max size of description is limited to 30 characters, so with the regular police it is a full line on the screen.
So it should be good in EDIT mode to print the description on the next line after "Description:", there is space

gullradriel, 2024-03-11:
Support is here for # comments. It's in the code and it's the very reason the line is shown in freqman, and in grey.

closed 2024-03-11 by gullradriel