portapack-mayhem/mayhem-firmware #1382: One-Button Radio Receiver Tuning

NotherNgineer, 2023-08-16:
Many radio receivers have quick-access memory buttons for tuning stored frequencies that the user may wish to listen to.  I propose an app that shows a "button" widget on the screen for each frequency stored in a specified "Freq List" file, allowing the user to simply press said button to switch between the stored frequencies (the mode and bandwidth would also change when the button is pressed).  As an example, each button could correspond to a local WFM radio station.

Currently in Mayhem firmware there are multiple steps required to switch frequencies when listening to an audio station.  For example, tuning a stored frequency in existing apps requires the following steps:
*  AUDIO app:  For every tuning change, click on the displayed frequency, then LOAD, then turn dial to a selected Freq List file ("CATEGORY"), then press the down button, then turn dial (or use keys) to a selected frequency in the file, then press Select twice, then change MODE if different.  (A lot of button pressing & dial turning)
*  SCANNER app:  Load a Freq List file, click <Pause> to stop scanning.  For all subsequent tuning changes, just turn the dial to select which stored frequency to listen to.
 
The original SCANNER app is my go-to simplest app for listening to WFM radio frequencies, but I would like the ability to press one button on the screen to jump between stored frequencies directly, as an alternative to turning the encoder dial.

This feature could be a separate app, or built into an existing audio app such as AUDIO, SCANNER, or RECON.  The app would also need to show the currently tuned frequency (perhaps by simply highlighting the selected button), and allow adjustment of RX gain and volume settings.  It should NOT have a spectrum display (to make room on the screen for the tuning buttons).  Optionally, holding the Select button when a button is highlighted might allow the button to be updated to a different frequency, like it works in some car radios (this would require updating the Freq List file).

sizzlik, 2023-08-30:
I like that idea. Fast switch is indeed cumbersome. A seperate app with a list of "customizable buttons" that start the wanted app with the given settings. Like having one list entry start the scanner with range x to y, another starts audio app at given freq and BW, the next starts capture app at given freq etc. That would be quite cool..but also much work i bet. I would welcome something in that direction.

NotherNgineer, 2024-12-29:
There's room for improvement but PR #2428 does what I was looking for so I'm closing this one.

closed 2024-12-29 by NotherNgineer