portapack-mayhem/mayhem-firmware #1389: Rfid and nfc

sizzlik, 2023-08-18:
I imagine that question came up often, but i didnt found good answers yet, so best to ask the most knowlegeable persons directly. Would it be possible (given you got or build the antenna for it) to read/write simple 125khz rfid chips or nfc cards? And i mean the really cheap ones that you get with an arduino nfc reader module..with no special protection. Im just reading into the topic and was wondering if an app could be made to play around in that area. Even stumbled across an old (2018) article from a sec reseacher who managed to read magnetic stripes using the havoc firmware. Would be interessting to use a near field probe on the hRF and read out these mini fobs or see what an nfc scanner sends, and maybe send something back. Just a wild thought..but maybe there is a posibility?

zxkmm, 2023-08-26:
> Would it be possible (given you got or build the antenna for it) to read/write simple 125khz rfid chips or nfc cards?   

it's possible *but hard* to do RFID things on a half duplex SDR. the best you are capable to do (with and half duplex SDR) is sniff and do mfkey32.
1. for hf cards: a legal RFID reader constantly read & write without lose (card) power supply, which half duplex can't seems do. as soon as you lose power (switch between Tx and Rx), the card powered off so it cause many attacks (or common read & write procedure) can't finish.  
2. for LF tags (which you mentioned):  it's out of the freq range, which hackrf support. and for a cheap SDR, the signal quality is too bad to do any RFID procedure, at that range.  

  
>And i mean the really cheap ones that you get with an arduino nfc reader module..    

it's meaningless to compare an SDR with an dedicated chip.

sizzlik, 2023-08-26:
Thanks for the detailed answer. I forgot that hackrf is not good at such low freqencys. Im just trying to learn the differences in chip types (man there are many). And maybe i should mention, im not looking for criminal stuff..there are ready build options..the thought just came up while fixing a keysy rfid duplicator wich is of not much use anymore other then playing around..at least ive not seen 125khz fobs anywhere anymore..all 13.56mhz. But let me ask another question, completly theoretical of course..a little brainfart. Hackrf got gpio pins wich portapak uses, are some still unused? If so, would it be possible for a tinkerer to hardware mod the pak, and connect a simple rfid module using i2c to 2 free gpio pins plus write an app for the firmware? Just theoretical..is it possible or what problems would occour? I know that may be a stupid question, but one cant learn without asking stupid questions. Thanks very much anyway.

zxkmm, 2023-08-27:
Hello, don't worry, what you asked is not a stupid question, plus I'm a noob as well, and everyone started as noob. so no need to worry at all. However, i'm not sure about the structure of this project, so idk if a RFID module is possible. Based on my knowledge it's possible to attach more module, because there is an illegal close-source fork of this repo already implemented GPS module, power management chip to battery level (through GPIO), compass and gyroscope on it. so i assume it's possible, but idk if it's through iic. 
However due to hardware performance, i think hardnested attack (for rfid it's important) isn't possible. maybe nrf52840 or some NXP chip can, but I'm not sure. maybe @cjheath or @pickeditmate knows better in this field.  
  
Maybe i should post this to some discussion zone instead of here, but since this thread is about discussion, so I assume maintainers are ok with it. :)

cjheath, 2023-08-27:
I think you have zero chance of making the HackRF work at 125kHz, let alone turning around Rx&Tx fast enough for RFID. It barely works at 1MHz.

zxkmm, 2023-08-27:
@cjheath:  wow how fast reply, ty!
he/she/they seems want to attach an RFID module on hackrf, so it could out of hackrf's radio part.

cjheath, 2023-08-27:
There seems little point is using the HackRF if you are going to add a new radio. Just add your radio to an Arduino or ESP32 and you're done

sizzlik, 2023-08-27:
it was just a bit brainfarting..sure i can use all the modules with all the MCU i collected, and as said, there are already toys to read rfid/nfc..that wasnt my point..i just like to tinker with stuff..just for that i got several hRF&PP...to fiddle with them and probably break in the process. Added a flat 6k mAh battery from a broken nanoVNA clone, speakers and mic, LED strips on the case that blink according to RX/TX led, external spf5189z amps, SPDT switch to switch between 2 antennas and as the hRF isnt good at LF i was wondering about possibilitys to add more to my frankenstein monster. In the end, Ossman made it as learning tool, and i learn everyday from all of you.
Guess this topic can be closed now..i was just curious. Thanks =)

closed 2023-08-27 by sizzlik