portapack-mayhem/mayhem-firmware #1390: firmware 1.7.0+ heats up the chip

mintertale, 2023-08-18:
**Describe the issue**
When you turn on this icon in the status bar to turn on the sound, the chip (with mark 4951 649A)  next to the speaker connector heats up.

**To Reproduce**
Steps to reproduce the behavior:
1. enable portapack with firmware 1.7.0+
2. enable this icon as on the screenshot
[image]

3. experience a hell of a heating of the chip (with mark 4951 649A)

**My Hardware**
HackRF One from 13 February 2014
Portapack h2 (?) doesn't look like any of these https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
any firmware where this sound activation icon appeared, on firmware 1.6.0 where it was not all worked fine and without heat

**Heated chip**
[image]

**My version Portapack H2**

[image]
[image]

NotherNgineer, 2023-08-18:
When using version 1.7.3+:
1. Are the symptoms the same regardless whether a headphone is plugged in?
2. Does the audio output work properly otherwise, e.g. when listening using the Audio app?

mintertale, 2023-08-18:
> When using version 1.7.3+:
> 
> 1. Are the symptoms the same regardless whether a headphone is plugged in?
> 2. Does the audio output work properly otherwise, e.g. when listening using the Audio app?

1. Yes, with this icon on, the problem is the same, but with headphones connected, the sound comes from both the earpiece and the speaker connected via the speaker port
2.Yes, everything works properly through Audio, as on previous firmware.

Also, I've now noticed that heating occurs when this icon and the speaker icon to the right of this icon are turned on. If I leave only the icon from the post and switch off the speaker icon, there is no heating and the sound switches from the speaker to the headphones if they are connected.

So in this state of icons everything works fine, no heating.
[image]

 Is there any way to make a setting so that portapack is always switched on in this state? Because when switching on they are both on and the chip is heating up.

NotherNgineer, 2023-08-18:
If the coin-cell battery is charged, note that the title-bar icon settings should persist across power-cycles.  (Settings may be reset during some firmware upgrades though)

You can also choose to disable any icon(s) that you don't want to use, but of course this isn't a good solution.

My PortaPack doesn't have the AK4951 chip so I can't test it myself, but I'll review the code to see if I can find anything.

mintertale, 2023-08-18:
> If the coin-cell battery is charged, note that the title-bar icon settings should persist across power-cycles. (Settings may be reset during some firmware upgrades though)
>
> You can also choose to [...]

Yes, the icon state is saved on reboot and can be used. 

I just used the latest firmware for some time and couldn't understand why the battery drains so fast and only after disassembling the case I noticed how much AK4951 warms up in the default icon settings.

NotherNgineer, 2023-08-18:
And if I understand correctly, the heating starts immediately after powering up, without even running any application?

mintertale, 2023-08-18:
> And if I understand correctly, the heating starts immediately after powering up, without even running any application?

yeah, immediately

NotherNgineer, 2023-08-18:
Can you try the following experiment for me using your AK4951 PortaPack?
1.  Flash f/w version 1.7.2.
2.  Under Settings -> User Interface, disable "Hide H1 Speaker option".  Speaker icon should appear on the title bar.
3.  Enable the Speaker output using the speaker icon.
4.  Run the Receive -> RadioSonde app.  This should unmute the speaker.
5.  See if the AK4951 chip heats up.

mintertale, 2023-08-18:
> Can you try the following experiment for me using your AK4951 PortaPack?
>
> 1. Flash f/w version 1.7.2.
> 2. Under Settings -> User Interface, disable "Hide H1 Speaker option".  Speaker icon should [...]

AK4951 starts to warm up as early as 3 points.

NotherNgineer, 2023-08-19:
If you have a multimeter, could you measure the voltage across the speaker contacts when the AK4951 is getting warm?  The AK4951 might possibly be outputting DC to the speaker coil even though it's silent.  Also is it a 4-ohm or 8-ohm speaker?  (you may be able to measure this when the PortaPack is off)

mintertale, 2023-08-19:
> If you have a multimeter, could you measure the DC voltage across the speaker contacts when the AK4951 is getting warm? The AK4951 might possibly be outputting DC to the speaker coil even though it's silent. Also is it a 4-ohm or 8-ohm speaker? (you may be able to measure this when the PortaPack is off)

voltage at the speacker contact when the speaker icon is switched on - 0 V. Speaker 2W 8 Ohm.

NotherNgineer, 2023-08-19:
According to the AK4951 specs, the chip shouldn't draw more than about 20mW regardless of configuration setting, IF the speaker & headphones are disconnected.  I don't suppose it would be easy to disconnect the speaker as a test?

I will try to think of some more things to try.  Meanwhile it would be good if someone else with an AK4951-based PortaPack could check if they're seeing the same issue.  (@Brumi-2021)

TLDR; enabling the AK4951 headphone output seems OK but when the Speaker is enabled the AK4951 consumes lots of power & gets hot even when no apps are running.  All firmware versions are probably affected, but 1.7.3 made it easier to enable the Speaker by always showing the speaker icon on the title bar by default.

mintertale, 2023-08-19:
> 

With the speaker icon off and the headphone icon on, everything works perfectly. If headphones are not connected - the speaker works, when headphones are connected, the headphones are switched on....
Now I will unsolder 1 pin of dynamin and try without it

mintertale, 2023-08-19:
> According to the AK4951 specs, the chip shouldn't draw more than about 20mW regardless of configuration setting, IF the speaker & headphones are disconnected. I don't suppose it would be easy to [...]

With the speaker icon off and the headphone icon on, everything works perfectly. If headphones are not connected - the speaker works, when headphones are connected, the headphones are switched on....
Now I will unsolder 1 pin of dynamin and try without it

mintertale, 2023-08-19:
@NotherNgineer Unsoldered 1 speaker wire and the problem still persists, AK4951 is still warming up

mintertale, 2023-08-19:
and for some reason when the cable is disconnected the icons state is not saved, every time it switches on by itself
The rechargeable battery is fully charged, the battery is still standing

Brumi-2021, 2023-08-19:
Hello @mintertale ,  
how do you detect that problem  , I mean checking the consumption ?   Hopefully no smoke ...., not broken the IC .... The warm of the AK IC is getting too hot > 50 or 60º (not possible to touch) or just warm , below 40 -45º ? 
I hope not broken,... I mean is it easy to detect with the closed case , or you just find out because you do not use any case box.

So far we only have H1 schmatic and we believe that H2 makers , just clone that schematic implementation part , (at least that was our assumption) 

Checking original H1 schmatic, as you could see, there is independent SP and HP outputs , totally isolated from the device itself . And by SW, following IC AK4951  specsheet ,  you can de-activate one or the other audio sound output independently . 
(that it is not possible to do it with H2 boards with WM8731 devices . that device  does not have independent HP / SP control , therefore in my H2+ the PCB tracks are just using SPK output , connected to HP in parallel ).

I tested @NotherNgineer  icon implentation and it was working perfect in my H2+ PCB 3.6 with AK4951  IC with SPK and HP , but I did not touch the device.    If necessary , I can re-assembly that special H2+ with AK board that I have as investigation spare. 

If your H2 boards really has also independent PCB tracks, like mine,  for SP / HP , please check carefully by visual inspection , all the PCB IC audio pin solders and surround SMD's , and SPK output and HP connector ,  to check that there is not any solder bridge .

[image]

NotherNgineer, 2023-08-19:
> and for some reason when the cable is disconnected the icons state is not saved, every time it switches on by itself The rechargeable battery is fully charged, the battery is still standing FW 1.7.4

I'm confused by what you're saying here.  Are you saying that persistent memory settings are not retained when the speaker wire is disconnected?  Note that all persistent memory settings will be reinitialized when you flash version 1.7.4, so perhaps that's what you're seeing?

mintertale, 2023-08-19:
> I'm confused by what you're saying here. Are you saying that persistent memory settings are not retained when the speaker wire is disconnected? Note that all persistent memory settings will be [...]

No, I mean after switching portapack off and on, the settings are not saved. But I just decided to check the DL1220 battery that is on the portapack and it has 0.02V. it's the first time I've seen such a low battery, maybe that's it.

https://github.com/eried/portapack-mayhem/assets/108941609/8c80f51c-b497-471b-8dc7-bcd2242bba60

mintertale, 2023-08-19:
Hi, @Brumi-2021 , 
Yes how, I actually went to clean the hackrf from dust (I carry it in its branded case from OpenSourceSDRLab and needed to listen to the frequency, I switched it on and accidentally touched the spot just to the right and above the cross buttons and almost burned my finger. The temperature is clearly over 60 degrees. 
In the video I posted above when I showed you that I don't save states after rebooting the portapack you can see how I even have the display brightness change when I switch the speaker icon on and off

Just before that I used 1.6.0 for a long time and there were no problems, including heating, then I upgraded to 1.7.2 and also seemed to be all right, but with the appearance of these icons with headphones and speaker on my portapack started problems with heat

NotherNgineer, 2023-08-19:
@mintertale Would you mind trying this same experiment again with 1.6.0 and 1.7.0 and let me know if/when it heats up?
1. Flash test f/w version.
2. Under Settings -> User Interface, disable "Hide H1 Speaker option". Speaker icon should appear on the title bar.
3. Enable the Speaker output using the speaker icon.
4. Run the Receive -> RadioSonde app. This should unmute the speaker.
5. See if the AK4951 chip heats up.

NotherNgineer, 2023-08-19:
@mintertale On my H2+R4 PortaPack (with WM8731 audio chip), I plugged the USB charging cable into a USB power meter, waited for the battery to be fully charged, then tried turning Mute on & off in multiple scenarios:  while idle, while running the Audio app, and while running the RadioSonde app.  Current consumption varied between apps, but toggling Mute on & off resulted in <10mA difference in all cases.  I know, I know, I don't have the AK4951 audio chip on my board, so this test didn't tell us much.

However, Brumi did a similar experiment on his H2+ with the AK4951 audio chip (he actually disconnected the LiPo battery), and he also reported that there was no measurable increase in current when turning the Speaker on/off or Mute on/off.

If you have a USB voltage/current meter, perhaps you could try the same experiment, to see how much current your AK4951 is actually consuming when its Speaker output is switched on.

Unless other users report the same issue, my guess is that there's a problem with your PortaPack board.  As suggested above, it would be good if you could also check if the problem shows up on your board now with 1.6.0 firmware WITH SPEAKER ENABLED as noted above.

Thank you for all your testing.

mintertale, 2023-08-20:
@NotherNgineer I checked firmware 1.6.0 and 1.7.0 as you wrote above. As a result, when I switch on the speaker icon, the chip AK4951 also heats up and when I enter radiosonde, it also heats up and does not stop heating up even after quitting this application and rebooting the device.

Anyway, it looks like my portapack is not fully charged yet and I decided to measure the voltage on the one I have.
With the speaker off, the current is 0.4 A.
With the speaker icon on and heating up, current 0.48 A

NotherNgineer, 2023-08-20:
You might try measuring the resistance of each of the speaker wires to ground, and checking closely the wires under that glue blob on the backside of the speaker connector.  I don't know for certain, but I am guessing that overheating could result if either of the speaker wires were shorted to ground, and you might still still get speaker sound but perhaps at half the expected volume level.  The schematic diagram shows ground on AK4951 pins 17, 25, and 29.  The diagram below might make it more clear which is pin 1.

[image]

Brumi-2021, 2023-08-20:
Hi @mintertale , all the current consumption measurements start to match . 

As Mark told you , I mounted my spare H2+ PCB v3 with AK audio IC , and disconnecting the Lipo battery and I measured around 200mA in idle mode , just booting and around 380 or 390mA in Audio App WFM . But I did not see any current jump of those 80mA that you measured with both green icons -with any icon combination (the icon works well muting independently the SPK or HP ,but no current change always same current . 
So I am still insisting that you may have any small solder bridge in your board . Check specially the SPK and HP output lines of the IC 

[image]
[image]

mintertale, 2023-08-20:
> You might try measuring the resistance of each of the speaker wires to ground, and checking closely the wires under that glue blob on the backside of the speaker connector. I don't know for certain, [...]

I checked the resistance of each speaker pin to ground - resistance 2 MOhm

mintertale, 2023-08-20:
Hello, @Brumi-2021 

I think I've checked everything, I can't find any tin bridges. I only soldered the wires to the speaker connector pins.
Here is my portapack without the speaker, the place near the speaker connector does not look good, just hot glue is difficult to clean completely, but the contacts I tested it and there are no shorts there.

Also without the speaker in this condition, the AK5941 still gets warm when the speaker icon is on

[image]
[image]

Brumi-2021, 2023-08-20:
Hi thanks @mintertale for sharing such a nice definition picture. 
so far , at first glance , solder visual inspection looks correct . Only two comments , but I guess not relevant at all . Pls double check  my red mark , (maybe just some glue resine and not real short circuit , then no impact at all ? ) my yellow mark , It seems a SMD mounted manually , not in normal horitzontal position? (But again if value is correct , and no visual short , it might be ok ) 
[image]

As you said , it might be that this PCB design is joining HP and SPK at some point … but it is really strange , because it was already separated in the original H1 PCB design. But in my other H2+ WM , they just extract headphone out from SPK , in parallel. 

Then , what will really help us , if some other user with your same PCB H2 Can test it and report to use , to see if that problem is general to that PCB design or just a faulty unit . 

Anyway , meanwhile we do not find out if your problem  is a general problem of that PCB design or not , I suggest  that you should specify  to @NotherNgineer which fw version was not creating that AK overheat problem , and we may need to apply it in our main branch .

mintertale, 2023-08-20:
> Hi thanks @mintertale for sharing such a nice definition picture.
> so far , at first glance , solder visual inspection looks correct . Only two comments , but I guess not relevant at all . Pls double [...]

Anyway, the yellow marked one is just glue, it must have gone snotty while I was removing it from the connector ;) there is an 8.8 kOhm resistor underneath.

And the one in red, yes it is a 10 kOhm resistor, but I also don't understand why it is standing sideways.... I can try to heat it up with a soldering iron and turn it properly, but I don't think it will help with the heating

f1ghy, 2023-08-20:
Hi, I had an issue near it, few mouth ago. One of the speaker wire was to the ground, or the speaker was conected from amplifier to the ground, or something like this. I dont remember exactly. That's append to me when i tred get the speaker swich off when plug in a headphone. from the jack swich. (Now I have a little swich for the speaker).
If this can help...
;o)

mintertale, 2023-08-20:
> Hi, I had an issue near it, few mouth ago. One of the speaker wire was to the ground, or the speaker was conected from amplifier to the ground, or something like this. I dont remember exactly. That's append to me when i tred get the speaker swich off when plug in a headphone from the jack swich. (Now I have a separate swich for the speaker).
> If this can help...
> ;o)

That's the thing, both contacts have a resistance greater than 2 Mohm when tested to ground.
And so yes, my speaker works perfectly well without any activating icons (before firmware 1.7.3) right after switching on and running apps, and when I connect headphones it automatically switches to headphones.
That's why it seems to me that this switch with the speaker icon is unnecessary in my portapack version.

Brumi-2021, 2023-08-20:
Hi reading your last comment , maybe your Portapack version just use the IC HP output lines and the headphone connector already switch HP to the SPK when not connected . (And maybe SPK out of the IC are not used or connected to GND by a capacitor . ) then if so, in your Portapack version , you do not need a fw switching version , because your hw  is already using the internal mechanical headphone plug switch .

mintertale, 2023-08-20:
> Hi reading your last comment , maybe your Portapack version just use the IC HP output lines and the headphone connector already switch HP to the SPK when not connected . (And maybe SPK out of the IC [...]

https://github.com/eried/portapack-mayhem/assets/108941609/a049f467-095d-4387-9aba-70be601eb3fc

Seems to be the case, I mentioned above in the comments that I have everything switching on its own as it is. I thought that was the normal way portapack works :)
And as I can understand from aliexpress, my portapack is the latest version from H2

One thing but, I currently have a dead 3V battery and can't replace it yet, I have to switch off the speaker icon every time...

It would be cool if it would be possible to change the settings for mayhem via a json file on a flash drive for example, and thus switch off the speaker and other parameters.

Brumi-2021, 2023-08-20:
From this video it is clear . You have already mechanical switch from the headphone plug connector .

As you have your cell button battery discharged I recommend you that you go to settings -> P. Memory Management and click “use sd card for persistent memory” and save it . Then it should remember your icon selections at any boot reading the sd card .

mintertale, 2023-08-20:
> From this video it is clear . You have already mechanical switch from the headphone plug connector .
>
> As you have your cell button battery discharged I recommend you that you go to settings -> P. [...]

Done, thanks for your help. In this case, I think I can close the ticket, but I need to leave comments for those who have the same portapack h2 version as me, so that they don't accidentally burn their AK4951 without even knowing about it (since 1.7.3 because of the enabled speaker icon right after flashing).

@NotherNgineer

NotherNgineer, 2023-08-20:
@mintertale  Glad that you and @Brumi-2021 figured it out.  I guess I should have asked you specifically to measure resistance from the AK4951's pins (19 & 20) to ground, since they're possibly shorted together or grounded on your board.  I didn't realize that you were able to hear the speaker when the icon was off, either, so the video helped.

I'm curious if you can read the part number of the 8-pin IC below the AK4951 -- is it by chance an 8122 amplifier?

mintertale, 2023-08-20:
> @mintertale Glad that you and @Brumi-2021 figured it out. I guess I should have asked you specifically to measure resistance from the AK4951's pins (19 & 20) to ground, since they're possibly shorted together or grounded on your board. I didn't realize that you were able to hear the speaker when the icon was off, either, so the video helped.
> 
> I'm curious if you can read the part number of the 8-pin IC below the AK4951 -- is it by chance an CS8122S / INS8002e amplifier?

19 & 20 have pin to ground:
4.17 kOhm and 0.17 kOhm

but this is only when the positive contact of the multimeter is at ground, when measuring backwards the resistance is infinity

When portapack off 19&20 are pulled to the ground

I tried to take as good a picture as possible, maybe you can tell what it is by the labelling
[image]

Brumi-2021, 2023-08-20:
Ok , glad to see that you keep persistent your icon selection! 

Then if I understood you well , you suggest us , that in next release , when detecting AK audio IC 

I-) The right SPK icon should be by default disabled (not green) . (Then , other user with  my HW, without that automatic mechanical switch may decide to  activate it and use it , because I find very annoying to plug the HP and not stop the independent SPK sound ) .

mintertale, 2023-08-20:
> Ok , glad to see that you keep persistent your icon selection!
>
> Then if I understood you well , you suggest us , that in next release , when detecting AK audio IC
>
> I-) The right SPK icon should be [...]

Yes, I think the right thing to do would be to default the right icon with the speaker off.

In my portapack version there is no sound on the speaker and headphones if only the right speaker icon is on.
If only the left icon with speaker and headphones is on, everything works correctly, there is automatic switching from speakers to headphones, etc.
If both icons are on, AK4951 starts to heat up and at low battery charge portapack stuck at all

NotherNgineer, 2023-08-20:
@mintertale   Ah yes that's the CS8122S amplifier chip, which can be used to differentiate your board model from the others that have the AK4951.  I can add it this model to the table in the Wiki, and I can also change the Speaker default in firmware to off.

I'd like to add photos of the front and back of your PortaPack board to the Wiki if you don't mind.  Should I use photos from above, or can you attach one each of the front and back, without circles, arrows, or fingerprints?  :-)

f1ghy, 2023-08-20:
Hi again, From the picture where The orange, and the green wire are pluged to the board. The green look to be connected to the center pin, so to the ground. try to connect it to the other side pin... the the IC should stop to over heat.
I think the midle pin is the ground. So the right connection to the speaker should be the 2 opposites pin !
I think so

f1ghy

mintertale, 2023-08-20:
@f1ghy Hello, agai, alas it's not like that, my AK4951 is warm even when the speaker is completely desoldered:)

@NotherNgineer I'm going to try to clean my portapack for a nice photo, if it doesn't work I'll find my revision on the internet and upload it to you.

f1ghy, 2023-08-20:
even when the speaker is completely desoldered:)... So Your right and i'm wrong, "\(-..°)/"
Sorry !
 ;o))

mintertale, 2023-08-20:
@NotherNgineer 

I tried very hard, but dust and hair from my two cats keep trying to sit on this black board, black is hard to photograph in general ;)

Use any of these photos:

[image]
[image]
[image]
[image]

NotherNgineer, 2023-08-20:
Thank you!

BTW, do you have any other issues with this OpenSourceSDRLab PortaPack H2 version or would you say it's completely compatible with Mayhem 1.7.4?

mintertale, 2023-08-20:
> Thank you!
> 
> BTW, do you have any other issues with this OpenSourceSDRLab PortaPack H2 version or would you say it's completely compatible with Mayhem 1.7.4?

I think everything else is working properly, haven't noticed any other problems.

Thank you for your help

NotherNgineer, 2023-08-20:
I've submitted PR #1396 to disable the AK4951 speaker by default whenever persistent memory is reset (or if coin cell battery is dead), on the assumption that there are more OpenSourceSDRLab H2's with this problem.

closed 2023-08-20 by NotherNgineer

GZ42159, 2023-10-15:
**Here my AK4951 chip overheats story:**
First: I didn't realize that turning off one of the SP/HP symbols would prevent the chip from overheating, I thought I had killed them by ESD or output shorting.
Second: I ordered and replaced the AK4951 chip (indeed not so easy with this chip size and without special tools) but still have the same problem with overheating.
**what now?** 
I got the technical data of the chips AK4951(24bit CODEC with MIC/HP/SPK-AMP) and CS8122S(3.0W mono Class-D audio amplifier) and studied a bit together with the hardware layout.
[https://www.digikey.co.uk/htmldatasheets/production/1745149/0/0/1/AK4951.pdf](url)
[http://www.chipstar-ic.com/UploadFiles/files/2111341821447.pdf](url)
Result: It looks like the hardware designer made a mistake and has connected the SP outputs of both chips AK4951 and CS8122S together, see schematic.
Such a solution/circuit can only work well if it is ensured that only one of the chips activates the output at any time. This will certainly not succeed by a software solution/initialization of the chips alone.
**My solution:**
Because the SP outputs of the AK4951(in my version of the PortaPackH2 board the CS8122S is responsible for this) are not used anyway, I simply disconnected them, see picture.
In my opinion (IMHO) this is the only safe solution to avoid heating the chips and without any drawback.
For all those who don't want to (or can't) do the separation of the lines on the PCB, there is only the software solution which shall ensure that never both chips activate the SP outputs at the same time.
By the way: It's amazing what high temperatures the AK1234 can withstand without being damaged.
What did I learn from the story?
Certainly a little more soldering and not immediately think that a chip is defective without precise anylysis :-)
Thanks to all supporters!

[image]
[image]
[image]

Brumi-2021, 2023-10-15:
Hi @GZ42159 , thanks for sharing it ! 
i also have an AK4951 Portapack (but it has not any additional audio amp ) and it has the HP outputs and SPK lines independent , that is why we could not reproduce that original problem . 

Anyway , thanks to @NotherNgineer we implemented the best sw countermeasure to avoid that hot temp , and high current drain in those special boards . 

In my case , just following original H1 design : 
[image]