portapack-mayhem/mayhem-firmware #1399: SSTV and Analog TV app

sizzlik, 2023-08-21:
Not really a feature request, just a question:
Im recently playing around with SSTV and try to send pictures and wondered..why is there a TX app but no RX app for SSTV? Is the analog tv app capable of recieving from the SSTV app? I try to send from one portapack to another and the analog tv app reacts to a signal but doesnt show anything. Is it even possible to send a pic from porta A to porta B? or is there a reason that there is no app to recieve SSTV? And what signal can the analog tv app handle? We still have a few tv stations you can recieve over dvb-t with just an antenna (PAL signal), but no luck as of yet when i tune into the freq they are supposed to be. Are those apps working and im just unlucky?
I would love to work on the wiki aswell and update it, when given the current state of some apps..some things are not easy to figure out if youre not a signal expert but just starting to learn SDR stuff. A more complete wiki would be quite helpfull i guess. Thanks anyways for all your hard work you guys..you dont get thanked enough..amazing work you do.

(I'll close this topic in a few days, even if unanswered..thanks again)

NotherNgineer, 2023-08-21:
I can only tell you that the Analog TV RX app expects PAL broadcast (wideband) TV signals, and I was recently fooling around with it trying to get NTSC to work.  Looks like there's no code yet in this app to detect sync pulses (between screens or lines), and the screen painting takes so long that the capture buffers stack up after the first screen image is painted (and then a bunch of partial images appear as capture buffers are lost).  Probably we need to add code to discard buffers while screen painting is in progress, and to detect the sync signals.  I presume that's why the app is still in the Yellow (more work needed) state.

Seems like narrow band SSTV could be supported by modifying the Analog TV app, but someone would need to contribute their time & knowledge to write the code...

sizzlik, 2023-08-24:
Got it..maybe someone got the knowlege and time to tweak the apps a bit someday..im looking forward for every nightly release. Even try to find my way into the source code but its not easy. Anyways..thanks for the answer and thanks again for all you guys work.

-topic closed as answered-

closed 2023-08-24 by sizzlik

NotherNgineer, 2023-08-24:
I have also updated the wiki to answer this question.  https://github.com/eried/portapack-mayhem/wiki/Analog-TV-Receiver