portapack-mayhem/mayhem-firmware #1400: Intermittent M0 Hard Fault in ADSB RX application

NotherNgineer, 2023-08-22:
Intermittent while running the ADSB app, sometimes when going in and out of the Geomap view.  Not sure what causes this, but I've seen it too.  Per @kallanreed, it’s in screen_pos() from screen_rect().  Affected firmware versions include 1.7.4.

Photo from Discord user "Lukes":
[image]

NotherNgineer, 2023-08-22:
The fault also occurs using f/w version 1.7.3:

[image]

NotherNgineer, 2023-08-22:
And with f/w version 1.7.2:

[image]

NotherNgineer, 2023-08-22:
And with 1.7.1 (just ruling out any recent code changes; looks like this problem has been around for awhile):

[image]

NotherNgineer, 2023-08-30:
Upon further analysis, it seems that there are two or three different types of faults when the "A/C details" screen is displayed after showing the map:

1)  Hard fault in ADSBRxView.  Call stack shows:  **on_tick_second -> updateDetailsandMap -> store_marker -> screen_rect -> screen_pos**.  Apparently an issue with trying to get the size from the parent rect pointer of the geomap (geomap is not actually being displayed at this time, and the call is coming from the timer tick ISR).  This issue goes away if either (a) the icao20.db file is renamed, or (b) the store_marker function is changed to use a hard-coded size such as 240x240 instead of trying to access the parent rect.

2)  SV#8 trap.  This one implies that an interrupt has occurred while already processing an interrupt.  Could be that interrupts are being incorrectly enabled inside an ISR (and maybe another timer tick comes in).  This might also occur in **on_tick_second -> updateDetailsandMap** when the "A/C details" are shown but it is **not solved by either of the changes above** so presumably it's a different issue.

3) Other (less frequent) M0 Guru Meditation traps, in the same test scenario.

kb8jnm, 2023-08-31:
I can confirm the error on my H2 as well.
However, I did not have it when I was using 1.7.4 and used a 128gb sd card.
Only after going to a smaller (32gb) SD card did it appear for me. Likely just lucky I guess since that makes no sense.
But I mention it in the case it has any relevance. Likely not. But not knowing, could it be related to access time of the SD ?
I will look for it and test again if I find it.

mtnbkr88, 2023-08-31:
As evidenced here https://www.facebook.com/groups/177623356165819/posts/1384172822177527/ , many of us are getting these hard faults. They happen within a few minutes in most apps for me in V1.7.4. Another power user observed they happen for him in all 1.7.X versions, but 1.6.X was okay for him. I hope they find and fix this issue.

NotherNgineer, 2023-08-31:
@mtnbkr88  If you are seeing this issue in other apps besides ADSB-RX, please submit an issue (or even add the details to this issue if you prefer).  An issue report greatly increases the likelihood of someone fixing it.

lucidliving69, 2023-09-11:
Happened to me today too

went to details then immediately to AC and it popped up 

running 1.7.4 stable

[image]

kylereedmsft, 2023-09-11:
Eesh, there's a credit card in the original image. 😱

closed 2023-10-11 by gullradriel