portapack-mayhem/mayhem-firmware #1406: Possibl to change tex encoding in POCSAQ

jonas18z, 2023-08-24:
Is it possible in some to be avlbe to change encoding for text in pocsaq? Seemlike swedish chars encode wrong and I would guess other special chartares

NotherNgineer, 2023-08-26:
Could you attach a screen-shot as an example?  If possible, please mark up the picture to note the characters that are displayed incorrectly and the character that you were expecting to see.

NotherNgineer, 2023-08-30:
I found more information on different pager character sets in this other project (multimon-ng):
https://github.com/EliasOenal/multimon-ng/issues/69

Seems like we could do some (optional) character swizzling if we have them in our font.

mfalkvidd, 2025-01-22:
> Seems like we could do some (optional) character swizzling if we have them in our font.

For Swedish, upper and lower case åäö (most common special chars) are available in the Fixed 8x16 font but not in the Fixed 5x8.