portapack-mayhem/mayhem-firmware #1411: Synthesizer not tuning correctly 2701 to 2900 range, incorrect frequency, OK in hackrf mode.

chris451gh, 2023-08-26:
----build: n230819

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 
When running portapack firmware the frequency range 2701 to 2950, in some tests tuning from above 2950 down will have the correct receive frequency, in some cases the wrong frequency is received, PLL lock problem or tuning scheme problem.

When running SDRAngel to control the portapack in "hackrf mode" there is correct tuning throughout the 2701 to 2950 range.
This was found while trying to receive an airport radar at 2710 mhz. A second Icom radio picked this up OK but the portapack appeared to be tuning near a 5G frequency. Sweeps indicate it is either locked on an incorrect frequency or tuning an incorrect frequency. 

**To Reproduce**
Steps to reproduce the behavior:
Provide a signal source that can tune between 2700 and 3000 mhz. This can be a sig gen or another hackRF run by external software such as SDRAngel or GnuRadio. 
1. Go to '.audio receive, spectrum mode, 2700mhz and verify signal from signal generator is centered...'
2. Tap on '.select frequency 1mhz steps 2700 and increment siggen and portapack up by 1mhz steps...'
3. Search upwards until signal appears again, somewhere between 2950 and 3000

**Expected behavior**
A clear and concise description of what you expected to happen.
Continous tuning between 2700 and 3000 on the correct frequency, no spurious crossover found near 2755 
**Affected versions**
Tested on n230819.
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.
Test setup:
Portapack with hackRFone and mayhem 
Signal generator, RF explorer on an antenna set to about 0dbm, set to 5 mhz steps and 2700-3000 sweeps.
Spectrum analyzer Latinex with 2700-3000 capability or a second HackRF one tuned to this range.
Reciever, 100khz to 3304 mhz. Icom R-20
This problem was discovered while attempting to receive airport radar at 2710 in Capture mode.
A 5G type signal appeared (wide band qam spectrum) where a radar was expected. The 5G appers to be in the 2.500 to 2.690 range.
Another portapack HackRFone in the Hackrf mode tuned this range correctly.
Boards: HackRFone and Portapack H2 with encoder knob replaced with US version. 
The preamp and post amp are verified working OK.
It appears to be a firmware bug as the HackRF mode works in this band.
This firmware displays the tuned frequency the amp gains and the lowpass, I note that some tuning frequencys generate a small offset in the last 6 digits on the blue background display. 
I will try earlier builds upon request, but I do recall trying to pick up the radars before with no luck on earlier builds.

Brumi-2021, 2023-08-26:
Hi @chris451gh , maybe that one may have also some relationship with that #553 . There l I also confirmed some strange behaviour in some 2.2 Ghz  freq range .

but if through HackRF mode , we can have a correct tune it , we should be able to tune correctly through standalone Portapack mayhem fw . 

Cheers

f1ghy, 2023-08-26:
Hi, I also notice this bug around ATV frequenc around 2.5 GHz on Portapack mode (not on HackRF/PC mode). Nothing is perfect ! 😭🙂

chris451gh, 2023-08-26:
I tried portapack at 2335 as a test. Tx and RX appear OK. There is no morse mod in the microphone mode. Audio from USB works. In "microphone mode' it does transmit and receive in all modes. 
In 'morse mode' off the main menu "fm modulation" a morse message is heard on the fm carrier.
In 'cw' mode there is just a carrier, does not seem to modulate.
This could be the 'dc carrier' that leaks through, but in cw mode no side tone is generated or routed to the modulation. I also tried 145.500mhz and it behaves the same way. That seems to be another bug, cw modulation is not routed but fm audio side tone is OK. I read 553 and my finding is a little different, that the side tone is missing, but it is tuning there with the mix carrier, but no side tone is fed into the carrier in cw mode.
I do see M0 guru meditation hard fault in this mode.
r0 1
r1 1
r2 1
r3 71
r12 f7
lr 13D35
pc 13d34
So the synthesizer seems OK in the 2300 range. I know there is a converter that switches around 2.75ghz so how this is handled in the frequency calculation and the settings for all oscillators may be the problem. I loaded SDRAngel onto an android phone,  I will try hackRF mode in the field running external control that way as a workaround and see what I get. 6 months ago or so the radar at the airport was tuned to a different frequency that I could receive with hackRF but they moved it back to the 2710 again where mayhem firmware has trouble tuning.

NotherNgineer, 2024-01-27:
PR #1629 might resolve the CW transmission issue.
PR #1819 may help with the tuning issues in some frequency ranges.

Please re-test with n_240128 when it comes out later tonight.

chris451gh, 2024-01-29:
I installed 240128 and Tested with two hackrf portapacks and an icom r20
and the fix appears to work. I will test it with a 2-4ghz sweeper and a
4-7ghz signal generator next week. Best test method is set portapack to
wide bandwidth with a waterfall and tune the portapack while sweeping and
observe the sweep signal for correct direction. There were mixing products
in that range where it switches between converters to the Iq dac adc and
direct to it. Inverted spurious mixes to be expected from the block
diagram. Thanks for finding the source of this problem!

gullradriel, 2024-04-06:
> I installed 240128 and Tested with two hackrf portapacks and an icom r20 and the fix appears to work.

Thanks for the reports and tests.

closed 2024-04-06 by gullradriel