portapack-mayhem/mayhem-firmware #1412: ~8KB wasted "fill" bytes in ROM read-only data segment

NotherNgineer, 2023-08-26:
Looking at the map file for the application code only (not baseband), the ARM GCC linker is aligning everything in the .rodata segment (read-only data & text string constants) to 16-byte boundaries, which I believe is unnecessary for most constant data.  For example, a 2-byte literal string such as "\n" (C/R followed by null) consumes 16 bytes of ROM space due to 14 unnecessary fill bytes.

I suspect this alignment could be reduced somehow for most constant data, which would save up to ~8KB of ROM space (based on ~1000 "fill" lines in the .rodata segment in the map file and an average ~8 bytes per "fill").

TBD how to do it...

bernd-herzog, 2023-09-14:
Thats a great idea. On how to do it:
In both
* firmware/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M0/ld/LPC43xx_M0.ld (for the large ui part)
* firmware/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4/ld/LPC43xx_M4.ld (for every baseband image)

you will find:

```
    .text : ALIGN(16) SUBALIGN(16)
    {
        *(.text.startup.*)
        *(.text)
        *(.text.*)
        *(.rodata)
        *(.rodata.*)
        *(.glue_7t)
        *(.glue_7)
        *(.gcc*)
    } > flash

```

gullradriel, 2023-10-07:
Solved by https://github.com/eried/portapack-mayhem/pull/1473

closed 2023-10-07 by gullradriel