portapack-mayhem/mayhem-firmware #1417: Capability to format sdcard

zxkmm, 2023-08-29:
**Is your feature request related to a problem? Please describe.**
It's not yet possible to format sdcard on portapack (without a PC).
**Describe the solution you'd like**
Implementing the real formatting sdcard (instead of filling random block to sdcard)
**Describe alternatives you've considered**
Implementing the *real* formatting is maybe tricky so a simple "clear" feature maybe an alternative solution.

**Additional**
If i didn't read the code wrong, currently the "wipe sdcard" button/app is only for writes random data into sdcard, so it's more like made for "emergency wiping evidence". it's useful in pentesting, so we can keep it, but a "real format" or at least "clear" feature is still important.  
this issue is an continuing of talk in [here](https://discord.com/channels/478685461456617484/478685461905276928/1146059304143687750).

sizzlik, 2023-08-30:
just out of curiousity..for what would one need to format the card with the hackrf? If you got no pc on hand and you format it with the hackrf, you got an empty hackrf..and then rebuild the directory structure and files by hand on the device? Im just wondering on what occations someone needs to format the card "out in the wild"?

zxkmm, 2023-08-30:
It’s a really nice question, ty for asking! Probably considered as an “init” tool, it is useful?

And yes about your question I’m thinking about a “clear” feature like I mentioned above, which like like “clear those files which is not from stock (our sdcard standard files/dirs).” This is really easy to implement but I’m preparing for an important exam these days so it would be so nice if someone can do it. (Maybe you? :P)

发自我的 iPhone