portapack-mayhem/mayhem-firmware #1421: Jammer menu button illegal 100k fine + jail accusation

cyb3rw0lf, 2023-08-30:
I want to bring to your attention that a friend of mine have been charged by the government's telecommunication department for up to 100k + jail for the accusation of possessing and importing a "Jammer" because the custom intercepted the shipment of his HackRF and saw the "Jammer" button in the portapack menu. Now he's discussing with lawyers and debating about the power and capacity of the jammer.

You might want to reconsider this button menu name at least.

sizzlik, 2023-08-30:
Isnt that the fault/problem of the buyer? If you import something thats not allowed in your country, you risk to get caught, and if caught its on you. Same for flipper zero, if its not allowed in your country and you get caught importing, you cant blame the makers and ask to change it. Here the posession is not illegal, jamming others stuff is..like Ham radios without license, youre allowed to have it and listen, just dont push the PTT button without license or you can get hefty fines.

cyb3rw0lf, 2023-08-30:
It's weird because there are official stores selling the hackrf in the country, so he bought it but from aliexpress as it was cheaper. You would think if they allow regular store to sell it, is legal, right? So it would seems they only check for import, but I wonder what happens if you have it in the backpack and the police stop you. Any SDR that can transmit could jam, but the fact that there's the button directly in the menu makes it think that this is one of the main feature. These are not technical people, for them if the law say no jam and you have jam button means you are guilty.
So the question is do we really need to have that direct button? It just create unnecessary troubles for SDR users. Can it be renamed "Fast Transmission" or "Bulk Transmission" or whatever.

maietta, 2023-08-30:
The hackRF and the software ecosystem are primarily used as research tools. There are some other use-cases for commercial applications and when SHTF, this could prove to be a life-saving feature when paired with a powerful RF amplifier tuned to the range you need to scramble. Like a bad plot in a silly action movie, a hackRF in theory could disrupt a terrorist's ability to remotely detonate a bomb under right circumstances, albeit unlikely.

The responsibility to adhere to the laws in each user's jurisdiction is on the user. Here in the United States, it's illegal to transmit without some sort of an armature radio license, with each license giving a scope of use. However, there are circumstances where those laws do not apply, such as an emergency situation.

Years ago,  use my handheld HAM radio to call for help directly to dispatch during a super dangerous situation, explained that I wasn't able to call 911 due to cellular outage. This could have possibly saved someone's life.

Sounds like your friend is completely irresponsible and needs to find a new hobby.

cyb3rw0lf, 2023-08-30:
> 

how is he irresponsible? maybe you didn't read the post

maietta, 2023-08-30:
> > 
> 
> how is he irresponsible? maybe you didn't read the post

Your "friend" was irresponsible for pushing buttons he without care. Don't go pushing buttons unless you know what they do. That applies to all things in the world, not just a research tool. Maybe we should turn the button bright red and place alarms all over it, but that might cause more people to be tempted to push the button out of curiousity.

At the end of the day, why did your "friend", have a hackRF with a portapack to begin with? Sounds to me like you are a fed, tbh.

maietta, 2023-08-30:
As far as I know, including from my own experience, the hackRF and portapack units aren't sold pre-assembled nor do they have the firmware preinstalled. I chose to buy my units from reputable sellers and these required assembly.

Your friend should contact the seller and let them know what happened, and work to have the advertising of the product include the warning at minimum, but also to perhaps ban the export of the product to that country, or have the seller remove the firmware prior to shipping.

Whatever the case may be, your friend shouldn't be facing any fines but it's not on us as developers to be told that we shouldn't include something because of the potential for abuse.

(I don't currently contribute to the project, but I have written software for the hackRF and intend to publish soon)

maietta, 2023-08-30:
> > 
> 
> how is he irresponsible? maybe you didn't read the post

Super likely I misread your post and yes, I misread it. I skim-read thousands of posts a day before I finish my first cup of coffee before my day is started.

sizzlik, 2023-08-30:
hehe..try to ask aliexpress to stop delivery to a certain country..doubt you got much luck there. And the ali version come indeed preloaded with mayhem and sometimes pre assembled. All of mine came preloaded with the latest stable version.

cyb3rw0lf, 2023-08-30:
> And if this project DID remove this feature, I would fork the project and re-add it. Others would too. Just in spite.

there's no need to remove the feature, just rename the button or have a toggle in the settings.

yes it was preloaded with 1.7.3

eried, 2023-08-30:
Is there is any implications for us? We don't really sell anything

sizzlik, 2023-08-30:
rename to strawberry jam hehe. Jokes aside, renaming or hiding it doesnt solve the problem. you never know what you recieve when ordering from ali..could be old firmware without the renaming, could be custom firmware like H3..in general, this is open source..what you do with it is up to you and if not allowed in your place..up to you to know the laws of your place.

sizzlik, 2023-08-30:
> Is there is any implications for us? We don't really sell anything

Highly doubt that.

cyb3rw0lf, 2023-08-30:
maybe he could fork the project, remove the feature and compile a new firmware, then tell the govt in the "new" SW version the feature is removed. Would someone be willing to help to do that?

NotherNgineer, 2023-08-30:
@cyb3rw0lf   If you fork and build your own firmware using the instructions in the wiki, you can disable menu items by removing or commenting out line(s) in ui_navigation.cpp.  Specifically you can disable jammer by deleting or commenting out this one line:

`//        {"Jammer", ui::Color::green(), &bitmap_icon_jammer, [&nav]() { nav.push<JammerView>(); }},`

jLynx, 2023-08-31:
So if it's to do with the import before he even got it. Are you saying customs powered on the device and had a play with it to try investigate it @cyb3rw0lf?

cyb3rw0lf, 2023-09-01:
> So if it's to do with the import before he even got it. Are you saying customs powered on the device and had a play with it to try investigate it @cyb3rw0lf?

Yes custom open the package and sent it for investigation, then they powered it on, went on transmission menu and saw the jammer button, took a picture of it and flag it as Jammer device. Then they filed a penal case in court and sent the communication. Took about 3-4 weeks

sizzlik, 2023-09-05:
i think the issue can be closed now?

gullradriel, 2023-09-09:
> maybe he could fork the project, remove the feature and compile a new firmware, then tell the govt in the "new" SW version the feature is removed. Would someone be willing to help to do that?

If he contact us via discord, no problem.

Checking the wiki would help to first get a building environment. Modifying icons and stuff after is easy and we will have no problem helping.

closed 2023-09-09 by gullradriel

wsaun-edh, 2023-11-05:
> > > 
> > 
> > how is he irresponsible? maybe you didn't read the post
> 
> Your "friend" was irresponsible for pushing buttons without care. Don't go pushing buttons unless you know what they do. That applies to all things in the world, not just a research tool. Maybe we should turn the button bright red and place alarms all over it, but that might cause more people to be tempted to push the button out of curiousity.
> 
> At the end of the day, why did your "friend", have a hackRF with a portapack to begin with? Sounds to me like you are a fed, tbh.
> 
> Also, your "friend" probably knew what they were doing, because pressing that button even for a little while, isn't likely to get the government involved-- it would take a bit more than that. The "friend" must have been taunting law enforcement or something by repeatedly pushing this button at close range to it's intended target.

Your reading comprehension needs work, they didn't press the jamming button, it was discovered at customs

wsaun-edh, 2023-11-05:
> > > 
> > 
> > how is he irresponsible? maybe you didn't read the post
> 
> Super likely I misread your post and yes, I misread it. I skim-read thousands of posts a day before I finish my first cup of coffee before my day is started.

🤣🤣🤣🤣 dude you're weird