portapack-mayhem/mayhem-firmware #1426: MIC App Tuning Step for USB/LSB

kb8jnm, 2023-08-31:
In the MIC app,
When the 'mode' is USB/LSB the tuning step should be 1khz not 5khz
Not a problem if direct entry, but to much to accurately rx a usb/lsb signal using tuning dial.
You go past the signal in one step, for they are only 2.0-2.4khz wide, making them unintelligible/unattainable.

5Khz step is fine for AM, FMn & FMw but ssb needs to be 1khz

NotherNgineer, 2023-08-31:
A better default would make sense I think.  Also not sure why Step doesn't seem to be configurable in the Mic app...

Additional comments/observations below:

In 1.7.4+ note that you can hold in the Select button on the Frequency field for a second until a digit turns blue, then use Left/Right select which digit you'd like to adjust, and then use the Encoder Dial to adjust any digit up/down by 1 to tune more precisely.  (Press Select again to exit this tuning mode)

"App Settings might also help:
1. If "Load App Settings" is disabled (under Settings->AppSettings), I'm seeing a default step of 25KHz regardless of "Mode".  (Agree this is a bad default for USB/LSB)
2. If "Load App Settings" is enabled, the "step=" value in Hz is read from the file SETTINGS/tx_mic.ini (you can edit this value manually if you like and it will be loaded when the Mic app is started.  (Also note that if "Save App Settings" is enabled it'll get updated in the .ini file when the MIC app exits.)

kb8jnm, 2023-08-31:
The select freq field trick does do the job. I was unaware. Did not see it in the docs.
Perhaps I will help with the docs when I am better familiar with these things.

A lot of settings in the ini file do not come forward in the app when saving/loading to the SD.
I have been unable to determine if it was something I was missing/fat fingering or a bug.
One would think more of the possible settings could be brought forward from file.
I did get it to load the default step, so it is working there.

Still trying to get my head around the source code to create some custom stuff, related to MIC app. It has a lot of potential.
Just my slow-poke learning I guess. Need a speed course LOL

Brumi-2021, 2023-09-03:
Hi @kb8jnm , sorry current Mic App does not have direct step freq GUI control , it uses the same as last previous preset in other applications . If you got to Audio Rx App and set up step to 10Hz or 50Hz and  you will have that fine tune needed in SSB reception to have a natural voice pitch . 

If needed , we can consider to add it inside Mic app.
Let us know if it works for you 

cheers

kallanreed, 2023-09-04:
One caveat to @Brumi-2021's message and a possible work around. If you have App Settings enabled, it will continue to use what it was last set at. You can edit the /SETTINGS/tx_mic.ini and edit the step size there.

kb8jnm, 2023-09-04:
> Hi @kb8jnm , sorry current Mic App does not have direct step freq GUI control , it uses the same as last previous preset in other applications . If you got to Audio Rx App and set up step to 10Hz or 50Hz then going to Mic App , you will have that fine tune step freq needed in SSB reception to have a natural voice pitch .
> 
> If needed , we can consider to add it inside Mic app. Let us know if it works for you
> 
> cheers

As I said earlier
The select freq field trick does do the job. I was unaware. Did not see it in the docs.
Refering to @NotherNgineer  post
In 1.7.4+ note that you can hold in the Select button on the Frequency field for a second until a digit turns blue, then use Left/Right select which digit you'd like to adjust, and then use the Encoder Dial to adjust any digit up/down by 1 to tune more precisely. (Press Select again to exit this tuning mode)

And Their seems to be an issue with reading writing to the app file elements I will make a new issue for.

So the work around above this is just fine. It just needs to be in the documentation for function.
Perhaps I missed it?
But it would be nice if all the apps worked in the same manor, whatever it is. A goal for 1.8.x ?

Brumi-2021, 2023-09-04:
Hi @kb8jnm ,
I also tested what @NotherNgineer and @kallanreed mentioned about editing /SETTINGS/tx_mic.ini and writing step : 10 Hz or 50 or 100Hz,  and it works very good for SSB pitch voice adj. 

And as @NotherNgineer mentioned you ," In 1.7.4+ note that you can hold in the Select button on the Frequency field for a second until a digit turns blue, then use Left/Right select which digit you'd like to adjust, and then use the Encoder Dial to adjust any digit up/down by 1 to tune more precisely. (Press Select again to exit this tuning mode)". 

I will add later  both explanations on the Mic App.
Let me know if with those two information and action would be enough to close that issue .

Cheers

kb8jnm, 2023-09-04:
Yes I agree the 'freq field trick' does the job as a work around.
And setting this in the mic.ini works as well. But One would not want to change the file to change the step between modes. So, The 'Selecting Freq Field Digit Trick' is the answer for changing 'step' between mode selection.

Yea, I guess you could release it as a bug. And It is now in the wiki so everyone will know or could know.

Brumi-2021, 2023-09-04:
Hi @kb8jnm , thanks for your feedback.
Ok, I just realised  now , that @NotherNgineer already added those comments inside the wiki  (Mic App) a few days ago. 
I also confirmed that he also added those 'freq. field trick' on top of the wiki, in the User Interface , (main controls) . Good !

Then , to be more explicit , I also added all those above comments (from @kallanreed and @NotherNgineer )  inside the wiki,  to the Audio Rx App , and Capture App.
Then let's close that issue ,   thanks to all ,

Brumi-2021, 2025-05-11:
Let's reopen that issue again , 
That PR, Adding 8.33KHz spacing #2628, break it again.