portapack-mayhem/mayhem-firmware #1432: Recon and Scanner UI locking up

X3r0C00L, 2023-09-03:
When running eithe recon or the scanner, if you begin to muck about with the controls such as changing back and forth between modulation types or adjusting other functions, they will randomly lock up requiring a reboot. 
This was on the official, so I updated the nightly last night and same problem, but now it will do the same in looking glass when making changes to modulation type but just less frequently.

sizzlik, 2023-09-04:
Cant reproduce the the problem. Tested on 2 devices, with scanner/looking glass and randomly switched settings around while scanning..got no hiccups.

X3r0C00L, 2023-09-05:
Ill definitely try a different SD card. When it locks up it mostly just locks the UI where it wont respond. Usually gets stuck on one frequency. Can still hear audio though. This just started for me on the last couple of updates. 
I did format the SD card as well when I updated to the nightly just to see if that would help. 

Ill run out and grab a new SD and get back with an update.

eried, 2023-09-06:
does not happen on my h2 either

gullradriel, 2023-09-09:
definitely can not reproduce, may be due to your hardware/sdcard.

closed 2023-09-09 by gullradriel