portapack-mayhem/mayhem-firmware #1441: Load a Freq. list from freqman into the pocsag reciever

sizzlik, 2023-09-09:
Currently in the POGSAC reciever app its only possible to manually enter the freq. or make steps with the encoder. It would be a nice addition to load the txt file from freqman into pogsac reciever and when loaded, turning the encoder will step through the list. There is already a pocsag list in the SD card freqman contents, but not used by the pogsac app.

And additional question..in the pogsac reciever app under configure, there is an option at the bottom to activate "beta". What is that option for or what does it do?

gullradriel, 2023-09-09:
I think you are wrong.
Have you just tried to focus on the frequency and press middle button ?
Because it's opening frequency manager and doing exactly what you are asking, plus you can also save.

Beta option is a new pocsag decoder under dev by @kallanreed . to activate it you have to check it, save, and restart the app. 

Use the latest bin in the PR if you want to try latest beta pocsag decoder: 
https://github.com/eried/portapack-mayhem/pull/1439

closed 2023-09-09 by gullradriel

sizzlik, 2023-09-09:
i am a blind idiot..closed.

gullradriel, 2023-09-09:
Don't be harsh on yourself. Maybe your question and the answer will help someone else (if they ever search in the issues, that is)
Cheers :-)

kallanreed, 2023-09-09:
Sounds like you want frequency field to behave like a scanner where it loads a list. That would be a feature ask. None of the other apps behave like this (aside from scanner types). It’s not a bad idea though.

sizzlik, 2023-09-09:
> Sounds like you want frequency field to behave like a scanner where it loads a list. That would be a feature ask. None of the other apps behave like this (aside from scanner types). It’s not a bad idea though.

Yepp..that was my train of thought. Load a complete list and with the encoder skip through it. I forgot that there is a save/load button in the manual entry field. Loading a list would be faster to switch between freqs, but thats just QoL stuff. Loading manually works just as fine.