portapack-mayhem/mayhem-firmware #1447: Don't boot after > 1.4.3 Upgrade. No Graphic driver seem's to work

ouilles, 2023-09-14:
Flashing Firmware from 1.5.0 to latest nightlybuild, won't boot.

Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot, without success.

**To Reproduce**
Steps to reproduce the behavior:
1. PortaPack with Mayhen Firmware 1.4.4
2. Mode HackRf On -> Yes
3. Blue screen With Sdr synoptic
4. With 1.7.4 archive, run flash_portapack_mayhem.bat

```
Devices detected: 0 DFU, 1 HackRF
Existing HackRF firmware version: local-57ed3eb
Erasing SPI flash...
Writing firmware to SPI flash...
Finished.
Please disconnect and reconnect your HackRF to run the new firmware.

If your device never boot after flashing, please refer to https://github.com/eried/portapack-mayhem/wiki/Won't-boot
```

6. Unplug, 5s, replug Usb : White screen flash, then blank screen, wait 10s, no changes
7. Unplug USB, 5s, Left touch pressed, replug USB : White screen, waiting more than 10s doesn't change anything
8. Unplug USB, 5s, Right touch pressed, replug USB : White screen, 3s then black screen, waiting more than 10s doesn't change anything
9. Unplug USB, 5s, Up touch pressed, replug USB : White screen flash, then black screen, waiting more than 10s doesn't change anything
10. Unplug USB, 5s, Down touch pressed, replug USB : White screen, waiting more than 10s doesn't change anything
11. Switch to dfu mode, 
12. run dfu_hackrf_one.bat -> Ok
13. run flash_portapack_mayhem.bat
14. Unplug USB, Plug USB, white screen 5s then Mayhem logo, everything is ok.

15. Tested with 1.31, 1.4.0, 1.4.3, 1.4.4 OK
16. Tested with 1.5.0, 1.5.1, 1.7.4, Nightly 230919 -> Fail White Blank Screen.

**My Hardware**
Portapack H1 - Blue - 20181029 - With touch screen

**Affected versions**
Version before 1.4.4 are Ok.
Starting from 1.5.0 to last nightly build, Flash ok, but white or blank screen.

**Were you able to update the firmware before?**
Yes

**Can you try the upgrade with a different PC/Portapack/HackRF?**
Only on HackRf Porta Pack, sorry.

----
Before creating this issue, **do the following**: 
**Done** * Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
**OK** * Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
**OK** * Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
**On Windows** * (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

jLynx, 2023-09-14:
Can you please remove your Rtc battery for 10 seconds then put it back in. The take a read of this for the latest nightly

ouilles, 2023-09-14:
I saw the fast flash of leds when doing numerous tests, but didn't found informations about this config menu.

It's a bit better, here are my tests :

1.  Battery out for 10 seconds, then in, portapack plug to hackrf card
2. Usb connected, screen flash white, then go black nothing after 10s
3. Press reset -> 3 leds flash
4. Press dfu -> Autodection (Tx Led Off)
5. press reset -> Screen Flash white then go black. Rx led flash quickly. Nothing on the screen
6. Press reset -> 3 leds flash
7. Press dfu twice -> Lcd driver 1(Tx On)
8. Press reset ->  Screen Flash white then go black nothing after 10s
9. Press reset -> 3 leds flash
10. Press dfu twice -> Lcd driver 2 (Tx blink fast)
11. Press reset  -> Lcd screen white nothing after 10s
12. Press reset -> 3 leds flash
13. Press dfu twice -> Lcd driver 2 QFP100 (Tx blink slow)
14. Press reset -> Lcd screen white nothing after 10s
15. Press reset -> 3 leds flash
16. Press dfu twice -> Lcd driver 1 QFP100 (Tx blink inverted slow)
17. **Press reset -> **Ok**, got Logo and seem's to be ok.**
18. Unplug Usb, 5s , Replug -> Lcd flash white, then go black, nothing new after 10s
19. Press reset -> 3 leds flash
20. Press dfu -> Always in Lcd blink inverted slow.
21. Press reset -> Lcd flash white, then go black, nothing new after 10s
22. Press reset -> 3 leds flash
23. Press dfu x 10 times -> Lcd driver 2 QFP100 (Tx led blink slow).
24. Press reset -> Lcd screen white, but nothing after 10s
25. Press reset -> 3 leds flash
26. Press dfu twice -> Lcd driver 1 QFP100 (Tx blink inverted slow)
27. **Press Reset -> Lcd screen White 2s, black 1s, and then It's Ok with the logo screen.**
28. Press reset -> Lcd screen flash white, then black, Rx led on. No more Logo.
29. Press reset -> 3 leds flash
30. Press dfu -> Tx blink inverted slow

Seem's like need to pass from LCD driver 2 QFP100 chip to LCD driver 1 QFP100 chip to work.

I've done some more tests :

1. Unplug usb 10s.
2. Plug usb with Left key pressed -> Lcd white
3. **Press reset twice while holding right key -> Lcd white, then black, And Logo screen.**

Module configured for Option 5: LCD driver 1 QFP100 chip (TX LED inverse blink slow)
1. Plug Usb -> black screen
2. Reset twice -> Black screen

BUT :

1. Plug Usb and key left pressed -> White screen
2. **Reset twice -> Ok**

I can do more tests if needed.

Brumi-2021, 2023-09-14:
Hello @ouilles ,  after reading quickly your post , 
I still do not have a clear image if your set is bricked or not . 

Did you follow the last part of this link ??? 

https://github.com/eried/portapack-mayhem/wiki/Won't-boot

ouilles, 2023-09-14:
Hello,

The module is not bricked. I can downgrade to 1.4.3 firmware using dfu mode.

Yes i have followed the last part of Won't boot (Special Step process (I) to recover it (usually only needs to be done just once)) without success.

To have the last nightly build working, i need to boot first with key left pressed and then reset with key right.

Key right alone doesn't work neither key left alone.

After a successfull boot, power off / power on cycle, the system doesn't boot correctly. Only after left reset /right reset.

jLynx, 2023-09-14:
@bernd-herzog maybe you will be able to help out here?

ouilles, 2023-09-14:
I'll try to find a small SD card to test.

But, Driver config is kept betwwen to power cycles (Tx Blink Slow Inverted).

Anyway, i'll test with an SDcard or changing the battery, and i'll give feedback.

ouilles, 2023-09-15:
Not backup battery problem, keep time clock with ok over 4 hours not connected.

SETTINGS -> P Mem Mgmt fail, but it's an other problem i think. SDCard seem's ok but not sure.

Brumi-2021, 2023-09-16:
When you said : “SETTINGS -> P Mem Mgmt fail, but it's an other problem i think. SDCard seem's ok but not sure.” , 
just in case , make sure also that you also activate Settings -> App Settings , both ticks Load / Save App settings , and later push “save” , and confirm that any selected User Interface settings are correctly recovered at any boot . 

Make sure that the SD CARD is working correctly , example test it , with any capture / replay , or activating splash ,….

jLynx, 2023-09-28:
If it boots correctly when you hold down the buttons, then the current code is OK. If you need to do it everytime, then your Rtc battery is flat (assuming you are not saving pmem file to SD card)

gullradriel, 2023-09-29:
For the pmem error your sdcard you may miss a SETTINGS directory at the root of it

ouilles, 2023-09-30:
You got it. Creating SETTINGS and now it's ok. Why does this directory is not on SD CARD image ?
Thank you.

ouilles, 2023-09-30:
> If it boots correctly when you hold down the buttons, then the current code is OK. If you need to do it everytime, then your Rtc battery is flat (assuming you are not saving pmem file to SD card)

It's not a battery problem. The time is correct even after an overnight stop.
To have a correct boot, you need a reset between two start-ups. Impossible to boot correctly otherwise.

ouilles, 2023-09-30:
I think we can close the issue. The sd card problems are solved. The boot remains, and there's a pull request associated with it.

closed 2023-09-30 by ouilles