portapack-mayhem/mayhem-firmware #1455: Looking glass "res" broken upgrading from 1.7.0 to 1.7.4

fdev31, 2023-09-20:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
Looking glass looks significantly slower than in previous versions.
"res" seem to have no effect in full range mode.
In today's nightly build it looks similar, but "res" seem to work better (low values leading to faster results). 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Looking glass'
2. Select 'FULL RANGE' preset
3. Check refresh speed (using F-SPCTR-V and the lowest RES value)

**Expected behavior**
The display refreshes

**Affected versions**
* Latest Stable release: The worst
* Latest Nightly release: Seem a tiny bit better, but maybe that's perception
* Previous working release: 1.7.0 (didn't test 1.7.1-3)

**Additional**
For some reason I'm not able to downgrade to make more tests...

gullradriel, 2023-10-11:
Unable to reproduce.
Have you tried latest nightly and a factory reset ?
Is your sdcard content up to date ?

fdev31, 2023-10-14:
Just installed the latest nightly, everything is up to date. I can confirm regardless of the values I set the scrolling speed is the same.
Before that it would impact the scrolling speed significantly, allowing to have a more rough but fast display.
I didn't see a "factory reset" option, how can I perform that?
(I thought each time I was flashing the firmware it would reset everything)

gullradriel, 2023-10-14:
From https://github.com/eried/portapack-mayhem/wiki
There is https://github.com/eried/portapack-mayhem/wiki/Factory-Defaults in Troubleshooting section.
I have some doubt it will help, but we never know.

gullradriel, 2023-10-14:
Additional information:
The speed of the waterfall and of the statistic updates is not controlled by the Glass.
Even 'res' is passed down in the radio API.
What you are describing is experiencing slower statistic updates than before. They are not false, there are less.
It may be due to the various radio modes evolution. I'm not sure we can do something about it.

fdev31, 2023-10-26:
Is it some kind of interrupt driven code ?
Does it set a flag or push a value to a queue ?

gullradriel, 2023-10-31:
It's event based. Statistics updates are broadcasted to an event message queue.
I'm closing the issue as we can not do anything about it and the 'res' is working.

closed 2023-10-31 by gullradriel