portapack-mayhem/mayhem-firmware #1461: Unstable portapack work

Alexitmisty, 2023-09-26:
----
Hello, I'm experimenting with different firmware and at the moment I have found problems for which I myself do not see a solution.
I bought portapack h2 and loaded firmware 1.7.4 and everything was good. So I built custom firmware without changes and loaded to portapack and here is problem. When I turn on looking glass and change frequency by wheel, portapack freezes and after some time I see this window.
[image]
[image]

I loaded some firmwares from discord and see same problem. I loaded nightly release and the problem remained. However if I flashing stable 1.7.4 everything works fine.
----
Below is a video of the protopack freezing and another error that popped up once:
https://github.com/eried/portapack-mayhem/assets/40298541/df51c48d-356b-4f90-8817-c86833da639f
[image]
----
I also added screenshots of building my custom firmware without changes. In fact this is a reset of stable 1.7.4. However, I still saw warnings:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
----
I would like to know if this is software or hardware problem? Did I buy a defective Protopak or is this a bug in the latest versions?

sizzlik, 2023-09-27:
Have you tried another SD card? Smells to me like a faulty sd card

Alexitmisty, 2023-09-27:
> Have you tried another SD card? Smells to me like a faulty sd card

One day I installed SD card. But now it’s not in the portapack.
[image]

Alexitmisty, 2023-09-27:
> Have you tried another SD card? Smells to me like a faulty sd card
I also checked on this MicroSD. MicroSD works good, portapack sees presets for looking glass, but portapack still freezes.
[image]

NotherNgineer, 2023-10-09:
I was concerned by the messed up screens in the first photos above (there should not be a red line in the middle of the stack dump screen!), but I attribute that to the Waterfall screen scrolling used in Looking Glass (and other apps) (related to open issue #1334).  That doesn't explain the Guru fault in the first place though.

Personally, I have success using Debian 11 and the old ARM 9 compiler version recommended in this Wiki (in case you might be using a different compiler):  https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

I see similar compiler warnings when building the submodules, so those are probably normal.

If you build from the 1.7.4 sources, do you get the same size binary image as in the official 1.7.4 release?

Test firmware images on discord could have issues.  Do the latest nightly release firmware images on github work for you?  https://github.com/eried/portapack-mayhem/releases

gullradriel, 2023-10-31:
No answers from OP.
Reopen an issue if you're still having the problem.

closed 2023-10-31 by gullradriel