portapack-mayhem/mayhem-firmware #1462: still blackscreen after tried everything

Jakz365, 2023-09-26:
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/eried/portapack-mayhem/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

sizzlik, 2023-09-27:
Nobody can help if you dont give any info beside "tried everything". When did you first encountered the blackscreen? Did you tried to roll back to a previous version of which you know it worked? Does the screen do anything like turn on backlight or short flicker to exclude the possibility of a hardware defect? Tried the button combination to load another display driver? Does the hackRF work without PP after flashing the standart original hRF firmware?

Jakz365, 2023-09-27:
Het sorry.    I hade wrote the whole story in an other groop, but What happend was that i by mistake/stupid choice to flash it by the utility so after tjat its completly blackscreen ive tried to flash by Computer cant find portapack even after i hold both dvu   &reset-reset-dvu for boot and so on i har version 1.7.4 i think i must had some jawbreaker file among firmwires as i remember

sizzlik, 2023-09-27:
So you fixed the issue? This is still pretty vague info. Did you try to flash it without the PP attached, back to original firmware from Greatscott? The dfu of the hRF alone is pretty much failsafe as far as i know. Flashing it back to standalone, then attaching the PP and flashing the mayhem fw might do the trick?

Jakz365, 2023-09-27:
Hmm can u tell me step by step ive maybe not tried that, i cant put it in rf mode bequise blackscreen so havent sucseed flashing it again

sizzlik, 2023-09-27:
remove the portapack from the hackrf, download the original firmware here: https://github.com/greatscottgadgets/hackrf/releases
then follow the instructions from here (or the readme file that comes with the original firmware):
https://hackrf.readthedocs.io/en/latest/updating_firmware.html (look at DFU Boot)
After that the hackrf should work as normal sdr.
Then you attach the Portapack again and flash the stable Mayhem Firmware again.

Jakz365, 2023-09-27:
My man thank u !

Jakz365, 2023-09-27:
Anything special i need to be extra carefull with or think about while doing it?

sizzlik, 2023-09-27:
Not really..get the firmware and the tools needed and follow the instructions in the docs. First try to SPI flash with the hackrf_spiflash program (part of the hackrf tools)..if that fails for some reason do a DFU boot and use dfu_utils to flash. The how to is all in the docs. Once you got it set back to the basic hackrf state, just put the portapack back on and flash the mayhem .bin file the same way you flashed the original .bin file before.

jLynx, 2024-01-15:
Closing due to being stale

closed 2024-01-15 by jLynx