portapack-mayhem/mayhem-firmware #1463: All types of modulation (FM/AM/SSB) in Soundboard / Morse / Signal gen

voxo22, 2023-09-27:
**Is your feature request related to a problem? Please describe.**
No.

It would be very helpful to have all types of modulation (not only FM) available to use in Soundboard, Morse and Signal Generator in Tx mode. Mic app has all of them. So I think it should not be very complicated to use them in other apps.
I often use hackrf as simple wobbler to matching up radio-TV stuff, FM/WFM modulation works very well, but AM/LSB/USB is missing (for matching-up LW/MW/SW bands). Hackrf w/ Portapack is even usable for matching-up LW radios, because Tx mode (compared to Rx mode) works well almost down to 0 kHz. Tested with Mic app.

daviderud, 2025-01-09:
Hi All,
I think it would be a useful feature, at least to have AM in addition to FM.

Cheers!

Robiii97, 2025-01-12:
Would also love to see this implemented in the near future.

daviderud, 2025-01-19:
The Signal Gen utility can be modified as follows (in proc_siggen.cpp):

```
                // // Do FM modulation
                // delta = sample * fm_delta;

                // phase += delta;
                // sphase = phase + (64 << 24);

                // re = (sine_table_i8[(sphase & 0xFF000000) >> 24]);  // sin LUT is not dealing with decimals , output range [-128 ,...127]
                // im = (sine_table_i8[(phase & 0xFF000000) >> 24]);

                // // Do Double Side Band modulation
                // re = sample;
                // im = 0;

                // // Do AM modulation (100% mod index)
                // re = (127 >> 1) + (sample >> 1);
                // im = 0;

                // Do AM modulation (50% mod index)
                re = 95 + (sample >> 2);
                im = 0;
```

By uncommenting/commenting the lines relevant to the modulation you want and recompiling, it works.

I am trying to put together the code with a modified user interface adding more Shapes, but there is not enough memory available. Have to think of a different implementation but I am not used to this code and need time. Idea would be to make the modulation a parameter configurable.

Current trials here: https://github.com/daviderud/mayhem-firmware/commit/ac165172d8b6b7ac43263f0c4ba9534e444a9e12