portapack-mayhem/mayhem-firmware #1468: ADS-B app shows no results if it's the first app executed after power-up

yellowz1234, 2023-09-30:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
I'm facing a rather annoying but hard to reproduce bug, where the ADS-B app will show no aircraft if it's the first app to be executed after power-up. This behavior is observed while multiple aircraft are visible in the sky.
Exiting the ADS-B app and immediately executing it again will correctly show all the visible aircraft.

The behavior cannot always be observed and sometimes it takes a few tries to reproduce.
Due to the random nature of the problem, I'm thinking it might be problem related to processor timing, uninitialized memory, etc.
The problem shows a lot of similarities to bug #1283 and PR #1285 .

If I run another receiver app before ADS-B, e.g. AIS, Audio or Search, the problem is not observed. 

I wasn't able to observe the same behavior in other receiver apps (like AIS).

Used equipment;
* Genuine HackRF R9
* Portapack H2+ R4

**To Reproduce**
Steps to reproduce the behavior:
1. Power up the Portapack.
2. Immediately go to "Receive" -> "ADS-B".
3. No aircraft will show up (even though they definitely should).
4. Exit the ADS-B app and immediately reopen it.
5. The aircraft will show up properly.

If the behavior cannot be observed, turn off/on portapack and try again.

**Expected behavior**
When the ADS-B app is the first app to run, it should show aircraft (if there are of course...).

**Affected versions**
* Problem observed with Mayhem 1.7.4 and every nightly after that.
* Have not tried with prior versions, but I believe it was working fine then.

**Additional**
Things I've tried, but made no difference:
* With or without SD card inserted.
* Different SD cards.
* Save/Load App Settings enabled or disabled.
* Clearing PMEM.
* Device charging or not charging.
* Different antennas.
* Pre-amp on/off. Different LNA/VGA settings.

Furthermore, the debug logs for successful and unsuccessful ADS-B sessions are exactly the same. 
Nothing unusual is observed when pressing the DFU button twice and looking at the receiver settings.

Brumi-2021, 2023-09-30:
Hello , I think it is related to a very old known problem (ghost signal) in the initial reception. I also saw it in Capture App. Long time ago ,  we  managed to solve it between applications , but recently it reappeared in the first application. 
I will have a look later , thanks

Brumi-2021, 2023-10-01:
Hi , @yellowz1234 ,
First I have checked my current local development PR (that is a branch of two weeks ago) , and it works fine , it does not show that past problem.

Then I recheck the  fw 1.7.4 and it shows a noisy reception in Capture , but still with good reception in  ADSB at first power up.

But then I checked yesterday's nightly   n_230930 and it works also well. 
Much better than fw 1.7.4 , quite clean Capture RX and ADSB has a nice reception at booting time or other second , third , ...,  launchments.
So I think we solved that problem , and now it seems all  ok . 
Pls. recheck again today's nightly 30th of Sept 23 and lets us know about it .

yellowz1234, 2023-10-02:
Hi @Brumi-2021 , thanks for taking the time to look into this issue.

I tried the nightly n_230930, but the problem is still there for ADS-B.
As you mentioned, the reception in Capture app looks cleaner, but the ADS-B app still shows no results sporadically.

At first I thought it might be related to a timing issue, i.e., how fast I launch the ADS-B app after power-up, as waiting a few moments before launching ADS-B increases the chance to show results more consistently. However, I also saw it occur a minute after booting the Portapack. 

I'm out of ideas at the moment, but If you think of any other things I should try, please let me know.

Brumi-2021, 2023-10-02:
Hi , thanks for your feedback,
I still think that current nightly is working well , at least I can not reproduce your problem. (In the past yes) .

Anyway , if you can , do that test : 
Once you got good ADSB reception , exit from ADSB App  and go to Capture app and make a screenshot of some of those nice random PAM pulses .  And compare them from first power up time screenshots when you use to have bad reception . 

In my case , now , with latest nightly , I do not see any difference . Now Capture seems clean in FM band or 1090 MHz . But In the past , I could clearly see , that when we had no planes reception in ADSB RX , 1090 MHz band and also FM band  appeared more noisy than in normal conditions . At that time , We got some adjacent channels intermodulations or interference . And that was a good reference to investigate and  fix it . 

If you could see a noticeable Capture screenshoot difference between first boot or second or third Capture  launch in 1090 MHz , pls  share and let us know .

yellowz1234, 2023-10-03:
Hi @Brumi-2021,

Below screenshots were taken using the latest nightly (n_231003).
- The image on the left was taken after running ADS-B app on first boot -> no results in ADS-B
- The image on the right was taken when running ADS-B for the second time -> ADS-B shows several results

[image]

The second image was less than 1 minute after the first. No other settings were changed.
They look fairly similar, but the first image looks a bit noisier to me. Not sure if this would explain the difference in ADS-B results though...

Brumi-2021, 2023-10-03:
Hi @yellowz1234 , yes indeed left side screen shows more noise and might be the reason . And you said that just waiting one 1min inside the Capture App , the noise disappears? Or in ADSB Rx , when is the first App called after booting , just waiting 1min , it starts to receive gradually more planes ? 

I will double check it , but so far in my case , I receive air planes just from the first seconds .

We need to understand if it is hardware related issue (due to some particular noisy DC switching regulator or any other noise coupling in your particular board  , or directly related to our latest fw.  
If you use old fw , that problem was not existing ? 
If so , pls let us know in which fw version was ok , and you can also compare in that fw , some capture screenshot just after booting or after the stabilisation time . 

To be honest , in the past we had "ghost " or adjacent channel interference , very easy to reproduce and check , and I could easily recognise in Capture App , but that problem it is gone in my unit .  

Other question , if you exit quickly the ADSB App in the first seconds and launch it again , then problem is gone ? 
And if so , do the same with Capture App , if just booting is noisy , what happens if quickly exit and launch Capture again  ? In that case , also the screenshot noise disappears?

yellowz1234, 2023-10-04:
Hi. 
Just a clarification to my previous post: I didn't wait inside the Capture app for 1 minute until the noise disappears.
I exited the app, launched ADS-B again and confirmed that the results show up correctly. After that, I open Capture again and the noise was gone. Sorry for the confusion.

Same thing for the ADS-B app: if it doesn't show any results on first try, I exit the app and immediately launch it again. It will always show the correct results on second, third and all consecutive tries. Failure only happens when ADS-B is run as the first app after boot.
Regardless of how long I wait inside the ADS-B app, if the results don"t show up immediately, they will never show up...

As you suggested, I went back to older firmwares, and confirmed the following behavior:
- 1.7.3: same problem as 1.7.4 and higher.
- 1.7.2: the problem did not occur. No matter how many times I tried, I could not make it fail and it always showed the correct results in ADS-B. Screenshots in the Capture app look very clean and do not show any noise.

I haven't tested all the nightlies between 1.7.2 and 1.7.3, but I think there must have been some changes between those versions that introduced this behavior.

Thanks.

Brumi-2021, 2023-10-04:
Hi @yellowz1234 , very clear your last post ! 

Then it reminds me very similar   behaviour as the previous "ghost  interference" that only appeared  in all Receicer apps , but at that time , it happened if we launched before any Tx App .(No ADSB Rx And Capture App noisy) . But we fixed it . (Now the problem is in the first direct  boot time -without need to go to any other TX app before ) . 

It is very good new,  that you could confirm that  this problem  was not there in fw  1.7 2 

I need to find the way to reproduce it , otherwise no way to investigate and solve it . 

How about other colleagues, can anyone also reproduce no ADSB RX just in the first try after booting ?

AC9DJ, 2023-10-04:
I'm running the 231004 nightly build and am seeing the same results as @yellow1234. I loaded 1.7.4 same problems observed. loaded 1.7.2 no problems noted. Anything specific you want tested?

Jim

Sent with [Proton Mail](https://proton.me/) secure email.

Brumi-2021, 2023-10-04:
Hi , good ,
it seems not only a unique set , it seems more general problem ,but I still can not reproduce in my H2+R4 , nor H1R2 , (but both with HackRF version <=6 )

Just in case , to better understand why I do not see that problem , which HackRF version are you running , the new one r9 or older same as me ? (If you go to Debug ->peripherals, ex. which max2837/39 do you see ?)

AC9DJ, 2023-10-05:
I'm showing MAX2837 on both of my devices.

Sent with [Proton Mail](https://proton.me/) secure email.

Brumi-2021, 2023-10-06:
Hi , till now I have tried to reproduce your problem  with H1 / H2+ , both with AK951 audio codec and original small QFP 64 CPLD. Just in case , this weekend I will try again , swapping the H2+ AK4951 with another spare PP H2+ WM 8731 Audio codec and big QFP100 CPLD. Let’s see .

AC9DJ, 2023-10-06:
Thanks Brumi-2021!

Appreciate your efforts.

Jim

Brumi-2021, 2023-10-07:
Hi @AC9DJ and @yellowz1234 , 
I just tested also with another spare PP : 
H2+R4 (wm8731 and big CPLD) with HackRF version 6 

and I regret to tell you that could neither  reproduce your problem.Here , testing fw (1.7.4 ,and  nightly n_23_10_3 ) both having a fast ADSB good reception , just booting . No difference from direct or second or third … time starting ADSB Rx App . 

I think , @yellowz1234 seems to have the new HackRF v9 , but I believe that you @AC9DJ are using a similar HackRF v6 than me , then it is very strange that I can no reproduce it ??? 
we will need more clues to be able to investigate it .

Cheers .

Brumi-2021, 2023-10-07:
Hello again , 
After playing with settings and changing fw from 1.7.4 to current nightly, I also could reproduce a no reception.
Maybe not related to your problem , but just in case let’s check it . 
When you got no reception , can you please press the button DFU two times and check Receiver debug parameters . I guess that maybe you have wrong “bandwith” = 1750000, and it should be minimum 2500000. 

Maybe in some intermediate version we had different pmem memory map , and it stored there a wrong value . 
if that is the case , you can go to the sd card and check in settings directory the ADSB_rx.ini file .
If it is there , you can check it’s content with the file manager , and if it wrong. Bandwidth, you can edit or directly delete that file , because it will be regenerated

yellowz1234, 2023-10-08:
Hello @Brumi-2021 . Thanks a lot for all the time you've spent investigating this issue. Really appreciated!

Indeed, I'm using the HackRF r9 with MAX2839.

I've checked the receiver settings during a failed ADS-B run, but they are all correct and the same as you pointed out.
Also, the contents of rx_adsb.ini is correct.

One thing I noticed during a failed ADS-B run is that the "M4 miss" count in Performance settings (press DFU button once) is much higher than during a successful run:
- failed run after boot: M4 miss = 33
- successful run on second launch: M4 miss = 3
Would this indicate something?

All other values (heap, stack, cpu%, etc.) are the same, but "M4 cpu%" is always very high, around 97%...

Brumi-2021, 2023-10-08:
Well ,just in case ,  it was worth to check your rx_adsb.ini settings . 

What we have also seen is that Audio Rx App shows just after booting in the first launch , very high M4 % usage 97 or 100% when it is selected  WFM demod and in some users it produces choppy audio sound . But it is recovered to the normal conditions and 56 % M4  after second or third... call . Maybe your problem might be related to it .
Later I will check if there is any M4% difference in ADSB Rx App from first booting call , to the next ones .

Brumi-2021, 2023-10-08:
Hi @yellowz1234 ,

Not sure why , but I realised that  anything compiled locally with gcc -arm-none-eabi version 10.3 , seems to have a lower  M4 % usage (86%)  in ADSB Rx  .  (if i got the official binary one compiled with 9.x it shows 98% usage and some MISSED M4 ). 

I just synchronized now the "next" repo , and this will be the same as tomorrow's nightly , but it is compiled locally with my  gcc 10.3 version.  And it shows to me in ADSB Rx App a  M4 usage 86% without any M4 MISSED ,  in booting or second , third time launch...
Pls. try just in case . If it works for you  , that might give us some  hint  investigation light .  (not the solution, but getting closer)

Note :  regarding Audio Rx in WFM mode , there is not any improvement , both compilers  9.x and 10.3 , shows now in both 100% usage . And this is something that we need also to investigate and  fix ...
[current_fw_next_8th_Oct.zip](https://github.com/eried/portapack-mayhem/files/12841421/current_fw_next_8th_Oct.zip)

yellowz1234, 2023-10-09:
Thank you, @Brumi-2021 .
I tried the provided test build, but it didn't solve the ADS-B issue for me.
In fact, with this build the issue happens every time when I boot the portapack. In previous versions after 1.7.2, it was sort of hit or miss, and the problem only happened randomly around half of the times. Now I get it consistently.

On the other hand, I confirmed that the M4 usage is round 86% with zero M4 missed in the test build.

NotherNgineer, 2023-10-10:
@yellowz1234  Since you are able to reproduce the issue, it might help if you could test the nightly builds between 1.7.2 and 1.7.3 to identify exactly which PR introduced the issue.  If you do this, I would suggest doing a binary search to reduce the number of nightly FW versions to test (e.g. start with a nightly build halfway between the working & failing versions).  That's what I would do if I could reproduce the issue.

yellowz1234, 2023-10-11:
Hi @NotherNgineer , excellent idea!

Having followed your suggestion, I can confirm with certainty that the problem starts from [nightly 2023-06-18](https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-06-18).

ADS-B RX works without fail in all nightlies up to and including 2023-06-17, resulting in strong reception and low M4 Miss count (<10).
However, from nightly 2023-06-18, ADS-B RX fails after power up almost every time and M4 Miss count is very high (~800). 

Hope this helps you guys pinpoint the issue. Thanks.

gullradriel, 2023-10-11:
May be related to #1160

NotherNgineer, 2023-10-11:
@gullradriel That's what I'd suspect too.

@yellowz1234  Did you actually test EVERY ONE of the nightly builds between 1.7.2 and 1.7.3?  I didn't expect that, but it would be terrific if you did, since bugs might come & go.

yellowz1234, 2023-10-11:
> @yellowz1234 Did you actually test EVERY ONE of the nightly builds between 1.7.2 and 1.7.3? I didn't expect that, but it would be terrific if you did, since bugs might come & go.

@NotherNgineer: I skipped a few versions here and there, but overall the behavior was pretty consistent that the errors only happened from nightly 2023-06-18 onward.

I also have the impression that earlier versions (before June 18) show more and quicker results, whereas it takes much longer for results to show up in later versions (after June 18) when they do work, and results are fewer...

NotherNgineer, 2023-10-11:
@gullradriel  I also made a change in #1161 to prevent SystemStatusView::refresh() from undoing the RX tuning offset (another possible culprit).

@yellowz1234  It might be interesting to experiment with modifying rx_adsb.ini to change the frequency to 1085000000 with Load App Settings enabled, to see if it matters.  (Using latest firmware)

Normally a tuning offset is used to avoid a "spike" at the desired center receive frequency.  In the case of a frequency of 1090000000 and sample rate of 2000000, the receiver will normally be programmed to 1090000000+500000 = 1095000000.  By reducing the frequency in the .ini file to 1085000000, the receiver will get programmed to the center frequency of 1090000000, as an experiment.

(I also noticed a separate issue #1487 with 0-size .ini files causing bogus bandwidth & sampling rate defaults, which could be causing RX issues for some people as well, but that doesn't seem to be the case above.)

NotherNgineer, 2023-10-11:
@yellowz1234  Nevermind the experiment above; looks like current firmware will ignore the frequency specified in the .ini file so the test won't work.

However, I am attaching a bootleg firmware version below that defeats the tuning offset by hard-coding the frequency to 1089500000 instead of 1090000000, resulting in the tuner being set to exactly 1090000000, just as an experiment.  Please try it and let me know if there's any difference.

(I also edited my comment above to correct frequency values BTW.)

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/12874938/portapack-h1_h2-mayhem.bin.zip)

yellowz1234, 2023-10-11:
@NotherNgineer I think you may be on to something here. 
Just tried the bootleg firmware and the "no reception on boot problem" happened only three times in 15+ tries. Previously I could reproduce more than half of the times, so there seems to be a difference.

Another thing I noticed with the test firmware is that the preamp needs to be enabled to receive any kind of ADS-B signal at all, and the signal seems to be much weaker. This was not the case with other firmwares.

NotherNgineer, 2023-10-11:
It doesn't sound like progress to me if the pre-amp always needs to be enabled with that last bootleg firmware!

Here's another bootleg below if you don't mind trying it.  This one reverts a minor CPLD change in PR #1160 (it does not have the ADSB tuning change from the bootleg above).  Theoretically it should have no effect, but let me know...

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/12875569/portapack-h1_h2-mayhem.bin.zip)

yellowz1234, 2023-10-12:
Hello @NotherNgineer , believe it or not, but the problem did not occur at all with your latest bootleg!
Solid ADS-B reception every single try and very low M4 miss count.
Whatever it is that you reverted, I think it solved the problem.

NotherNgineer, 2023-10-12:
@yellowz1234  Thank you for your testing -- that's great to hear.  Can you try another experiment for me (using this same bootleg)?  Please do a Transmit operation (using any of the Transmit apps) and then run ADSB-RX again afterwards.  Specifically I would like to know if the ADS-B reception remains good after a Transmit operation, or if a Transmit operation temporarily puts ADSB-RX into the problem state again.

@gullradriel @kallanreed @Brumi-2021  The test firmware above that seems to resolve the ADSB reception issue has the following change, effectively preventing hackrf::cpld::load_sram_no_verify() from being called unnecessarily on the first call to set_direction().  FYI, in case you have any theories as to why this works...
`   static rf::Direction direction{rf::Direction::Receive};    // changed from -1 [invalid]
`

yellowz1234, 2023-10-12:
@NotherNgineer After doing a transmit (e.g. POCSAG TX), ADS-B RX is not working. Also, Audio RX is not working and waterfall shows up empty.
Rebooting the portapack makes it work again.

NotherNgineer, 2023-10-12:
@yellowz1234  Can you please check if the RX issue after Transmit shows up in the latest nightly release and in 1.7.4 firmware as well (just to make sure it's not something funny in my test version)?  I presume this issue has existed since the 6/18 nightly as well.

Brumi-2021, 2023-10-12:
Hi @NotherNgineer , thanks for continuing investigating that ADSB boot problem (in some devices). 

long time ago , in my previous “ghost Rx interference” investigation , I introduced that call to 
hackrf::cpld::load_sram_no_verify(), at the exit of any Tx App , and that solved that “ghost interf “ reception , after using any TX App. 

But , I am not sure , if we need  to call it at booting time ..
Anyway ,  
I know that HackRF team , modified and improved long time ago the CPLD direction change . 
I wonder , if those two users that have that particular problem (and we can not reproduce ) may solve their problem if they update the Xilinx CPLD fw (that is in HackRF brd) . (My new theory , maybe they still have old CPLD fw , not updated to the final ) . 
(I will try to find a link to explain the steps )

yellowz1234, 2023-10-13:
> @yellowz1234 Can you please check if the RX issue after Transmit shows up in the latest nightly release and in 1.7.4 firmware as well (just to make sure it's not something funny in my test version)? I presume this issue has existed since the 6/18 nightly as well.

@NotherNgineer I confirmed that the RX issue after Transmit shows up in the latest nightly release (n_231013) and in 1.7.4 firmware as well.

Brumi-2021, 2023-10-13:
Hi @yellowz1234 and @AC9DJ ,
Based on the latest post discussions , I think it has meaning that you also try to reflash the Xilinx HackRF CPLD with latest binary stream . Doing that , at least we will discard or not that possible difference .

as you could see in the #1070 , another r9 user could solve his strange problems , just updating the CPLD .

[image]

if you can , please try it and let us know .
Personally, I think , big possibilities to fix your boot problem . Because I know that HackRF group improved and applied several PR’s about CPLD switching direction TX/Rx

AC9DJ, 2023-10-13:
Sorry haven't chimed in for a bit. I tried updating the CPDL and got this, "hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf Failed to open file: firmware/cpld/sgpio_if/default.xsvf"
Not sure if I missed something.

Thanks for all the help with this!

Jim

Sent with [Proton Mail](https://proton.me/) secure email.

AC9DJ, 2023-10-13:
Nevermind, I didn't have the full path in the command. Face palm.
ADS-B is working flawless for me every time on first start. I really appreciate the assistance.

Jim

James P. Anderson
cell: +1 262-287-2619
***@***.***

Sent with [Proton Mail](https://proton.me/) secure email.

Brumi-2021, 2023-10-13:
Hi @AC9DJ ,  
then you finally could update the CPLD binary and now it works well each first start ?
Or how did you fix your problem ? 

Cheers,

AC9DJ, 2023-10-13:
I updated the CPDL and ADS-B is working every first boot. That's all I did to fix the problem. Sounds like CPDL was out of date on my older board. Curious what changed after 1.7.2 that led to this. Just very happy it's working now.

James P. Anderson
cell: +1 262-287-2619
***@***.***

Sent with [Proton Mail](https://proton.me/) secure email.

Brumi-2021, 2023-10-13:
Very Glad and Happy ,  to hear that we fixed your problem !!!!
I hope that we will fix as well the same problem on @yellowz1234 's   device .

It seems quite popular , that some makers , are using very recent hardware platform, (including  r9) , but they are still recording very old  CPLD binary stream  , out of date (<2021) ,  It happened also to me in my H2+  r6,  and to some other r9 user's. 

@yellowz1234 ,  please also try , and if you also solve it , we will close that issue .

Cheers.

yellowz1234, 2023-10-14:
@Brumi-2021, @NotherNgineer Thanks for the many suggestions.
I updated the CPLD as per the provided instructions and did notice some different behavior.
(just wondering why the update was necessary, as I have the latest GreatScottGadgets r9 version from NooElec...)

Anyway, the behavior I observed was the following:

Using the latest bootleg version by @NotherNgineer:
- ADS-B RX on first boot almost always worked and showed solid results every time. If I remember correctly, I only had 1 failure out of 20+ tries.
- ADS-B RX or Audio RX after running POCSAG TX on boot work fine, however, ADS-B shows fewer results.
- The following scenario did not work: ADS-B RX after boot, followed by POCSAG TX, followed again by ADS-B RX -> no results in ADS-B.

On the other hand, using 1.7.4 or the n_231013 nightly, the behavior of ADS-B reception after boot did not improve at all.

This leads me to believe that the only thing the CPLD update improved is the performance of RX apps after running TX on boot.

@AC9DJ Which firmware version did you try? 
Can you try running ADS-B RX after boot, immediately followed by POCSAG TX, followed again by ADS-B RX?

Brumi-2021, 2023-10-14:
Hi @yellowz1234 , thanks for your extended test !
🤔,  but not so happy with it . I expected solving it . 
A few comments on it . NooElec is an excellent brand , I have also several saw filters from them , and always very good quality. But just in case , we neeed to be sure that we are testing different fw’s in similar full package condicitions, including CPLD Xilinx (except r6 / r9 platform hw  difference).

Now at least , we have same consistent results among r6 platform (@AC9DJ , @NotherNgineer , and myself ).

Regarding your different test results , again I am confused . Let’s see @AC9DJ answer to your question , and later I may need to discuss with @NotherNgineer .
Thanks

closed 2023-10-14 by gullradriel

AC9DJ, 2023-10-14:
> Hi ***@***.***(https://github.com/yellowz1234) , thanks for your extended test !
> Now at least , we have same consistent results among r6 platform ***@***.***(https://github.com/AC9DJ) , ***@***.***(https://github.com/NotherNgineer) , and myself ).
> Regarding your different test results , again I am confused . Let’s see ***@***.***(https://github.com/AC9DJ) answer to your question , and later I may need to discuss with ***@***.***(https://github.com/NotherNgineer) .

I followed what yellowz1234 wanted me to test and everything is working flawlessly for me. I’m actually seeing more contacts in ADS-B than I ever have. POCSAG is working flawlessly as well. Wish I could help out more in reproducing what you’re seeing. If you need me to do any further testing let me know.

Jim

NotherNgineer, 2023-10-14:
@yellowz1234  Although the original concern above was resolved by #1494, we still need to understand why RX isn't working after TX on your Portapack.  So more questions...

You mentioned that the following scenario did not work:  "ADS-B RX after boot, followed by POCSAG TX, followed again by ADS-B RX -> no results in ADS-B."

1.  If you follow that procedure, and then run ADS-B RX again afterwards, does ADS-B ever start working (or is a power cycle actually required to get RX working again)?

2.  Can you confirm that this test passes using firmware prior to the 6/18/23 nightly?

AC9DJ, 2023-10-15:
@yellowz1234,

I just tested again running ADS-B RX, the POCSAG TX, then ADS-B using the n231015 build with no issues.

Jim

James P. Anderson
***@***.***

Sent with [Proton Mail](https://proton.me/) secure email.

yellowz1234, 2023-10-15:
@NotherNgineer Good questions. I've noticed the following behavior:

- On a fresh 1.7.2 install (using "flash_portapack_mayhem.bat"), with or without having explicitly cleared Pmem in settings afterwards, the test passes every time without failing.
- After then flashing your second bootleg (also using "flash_portapack_mayhem.bat", without explicitly clearing Pmem in settings afterwards), the test passes every time as well.
- However, if I then clear Pmem settings, the test fails every time.

Could it be related to some Pmem settings that were introduced after n_230618?

If I run ADS-B RX another time after the failed attempt, it always works again. A power cycle is not necessary to get it working again.

@AC9DJ Glad it works for you now. Thanks for your support.

NotherNgineer, 2023-10-15:
@yellowz1234  When you say "the test" fails, I will assume that you're referring to the most recent test where you're running ADS-B RX right after POCSAG TX.  Note there is a "Debug -> Debug Dump" application that dumps Pmem contents to a file.  It might be interesting to run the Debug Dump app in the working and non-working scenarios above to check if there are any differences in Pmem configuration.  I would "expect" Pmem to be reset to the exact same defaults after a 1.7.2 -> n_231015 firmware upgrade, same as if you reset Pmem manually under Settings -> P Memory Management.  But we probably should check...

yellowz1234, 2023-10-15:
@NotherNgineer Indeed, with "the test" I meant "running ADS-B RX followed by POCSAG TX followed by ADS-B RX again".
I wasn't able to do a Pmem dump in the working scenario, as that function was not yet implemented in firmware 1.7.2  or nighly n_230617.

Brumi-2021, 2023-10-15:
Hello @yellowz1234 ,
Sometimes I  also got wrong behaviour and incorrect decoded audio ,when  changing fw versions (between 1.7.4 and latest ) 

Then , once you got, that the ADSB Rx test fails every time with latest version. n_231015, then , pls try to go Utilitilies -> File Manager , go inside SETTINGS holder and delete rx_adsb.ini . (Don’t worry , it will be generated again , in next ADSB Rx App call . )

That process , use to  fix me those kind of  pmem problems .  later , we can also check , which fw reset pmem breaks it , but we also need to find a way to solve your problem .

NotherNgineer, 2023-10-15:
> Indeed, with "the test" I meant "running ADS-B RX followed by POCSAG TX followed by ADS-B RX again".

I'm wondering if the RX after TX problem should be tracked in a separate issue ticket (since this one was closed).

> I wasn't able to do a Pmem dump in the working scenario, as that function was not yet implemented in firmware 1.7.2 or nighly n_230617.

For the "good" case, I guess I was thinking of n_231015 firmware but Before explicitly clearing Pmem.

yellowz1234, 2023-10-17:
@NotherNgineer Strangely enough, the Pmem dump files are the same before and after explicitly clearing Pmem on n_231015.

Some new behavior that I observed:
- Nightly n_231015: mostly working OK, behavior close to the bootleg firmware you provided.
- Nightly n_231016/n_231017: ADS-B RX behavior worse: never working on first try, sometimes working on second try or third try. Success rate is low.
- 1.8.0: no ADS-B RX at all, regardless how many times I tried or rebooted. I could also not get AIS RX to work and audio RX (FM stations) signals were very weak and noisy.

Did anything change after n_231015 that could cause this behavior?

I will open a new thread for this if you think it is better.