portapack-mayhem/mayhem-firmware #1472: Error

samukas81, 2023-10-06:
Error DFU in gps sim
Last firmware

sizzlik, 2023-10-06:
Bit more explicit?

samukas81, 2023-10-06:
When I choose the .c8 file for gps sim it gives an error and asks me to press the dfu button

sizzlik, 2023-10-06:
how did you create the .c8 file?

NotherNgineer, 2023-10-08:
I reproduced this with a .C8 file that was working earlier.  Displayed Guru fault message is "_parent is null".

samukas81, 2023-10-08:
exactly!!

NotherNgineer, 2023-10-09:
Looks like this issue occurs in the n_230928 nightly build or later, pointing to PR #1460.

samukas81, 2023-10-09:
Is it possible to fix this problem?

NotherNgineer, 2023-10-09:
"Bugs are free; fixes cost $$$ USD."  :-)

The fault is being triggered by the code line (someone might know exactly why already but I'll need to study it further):
`    button_play.focus();`

samukas81, 2023-10-09:
I believe in you programmers, you are geniuses!!

PR #1477 (2023-10-09): Kludge fix for null parent() in GPS Sim app
Fixes #1472 using the same kludge fix as in the PlayList app.  Not sure how #1460 triggered this issue to show up.

Caveat:  Focus doesn't get changed to the "Play" button automatically, so user will need to move focus manually.

@kallanreed, FYI
closed 2023-10-09 by gullradriel