portapack-mayhem/mayhem-firmware #1486: Weather Station APP (please)

DavidIgor, 2023-10-11:
app similar to this: 
https://github.com/flipperdevices/flipperzero-firmware/issues/2122
would be appreciated !

NotherNgineer, 2023-10-15:
The flipper issue 2122 mentioned above is regarding displaying temperatures in F vs C?

From the title, I assume the request here is to add support for decoding & displaying the data transmitted by various brands of wireless outdoor weather sensors (La Crosse, Acurite, Taylor, Ecowitt, etc).

Perhaps it would be possible to hack the TPMS app as a starting point.

DavidIgor, 2023-10-15:
oh sorry my bad (that flipper link) ... and yes it would be perfect to duplicate hacked tpms app

z-l-p, 2023-10-18:
The venerable [RTL_433](https://github.com/merbanan/rtl_433/) does this (for computers and SDRs) and [this open issue](https://github.com/eried/portapack-mayhem/issues/113) is a request to port RTL_433 decoders to Portapack.

DavidIgor, 2023-11-29:
already installed! will be testing it out...  good stuff! 
THANKS