portapack-mayhem/mayhem-firmware #1487: App Settings issue - incorrect BW & Sampling Rates due to empty .ini file

NotherNgineer, 2023-10-11:
When Save App Settings is DISABLED, a 0-size .ini file is being created in the SETTINGS folder whenever an app is closed.

If Load App Settings is subsequently enabled, default values are not selected correctly when the empty .ini file is read.

To reproduce:
-    Disable Load & Save App Settings in Settings menu.
-    Run ADSB-RX app, then exit.
-    Run FileMan and see a 0-size rx_adsb.ini file in the SETTINGS folder.
-    Re-enable Load & Save App Settings.
-    Run ADSB-RX app again, and use DFU to view incorrect default Bandwidth & Sampling Rate (which does not pick up any aircraft BTW).

(I don't think it should matter, but my coin cell battery is also dead so I'm using SD for PMEM)

[image]

[image]

NotherNgineer, 2023-10-11:
Adding @kallanreed - FYI

kallanreed, 2023-10-11:
This looks like it’s working by design? The defaults are set in the app setting struct and they aren’t overwritten by values from the file. The problem is that the defaults are bad for the app (radio_state is used to set correct values). There used to be a TODO in app_settings.cpp about keeping track of which values were actually read and only copy them to the radio model in that case.

NotherNgineer, 2023-10-11:
I've proposed a quick workaround for this in #1490, i.e. to pretend the file doesn't exist when it doesn't contain any valid app settings.

closed 2023-10-14 by NotherNgineer