portapack-mayhem/mayhem-firmware #1493: [Help] Booting an H2 without a battery

ruilvo, 2023-10-12:
I have a [PortaPack H2](https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions#h2).

It had an internal battery, but the battery stopped holding change and is now just a pillow.
However, if I remove it, the device doesn't boot. 

Is there a way to run the H2 without the internal battery? I don't mind running from a USB power bank.

Thanks.

If it's too much off-topic, just close this.

sizzlik, 2023-10-13:
As you said..simple powerbank wil do. Ofc the device wont boot with the battery removed and no other power source =) 
If you want a replacement batt, you can get them for cheap on aliexpress. Just search for "104050" or "BLY 104050" with 2500mAh...more mAh might be too thick to fit sandwiched between hRF and PP.

ruilvo, 2023-10-13:
The weird thing is that mine doesn't boot without a battery connected.
Or at least I don't think it does (?). Maybe it broke in the meanwhile? haha
But yeah, I'll get a battery and check, thanks.

closed 2023-10-13 by ruilvo

sizzlik, 2023-10-13:
To test i just removed the battery from my H2, boots perfectly fine when usb is connected. Dont think there is a difference in H2 and H2+ when it comes to battery related stuff.

ruilvo, 2023-10-13:
On mine the LED on the charging controller blinks extremely fast, and doesn't boot :/

sizzlik, 2023-10-13:
that doesnt sound good. Maybe join the discord and ask your question there aswell to check with other H2 owners?

ruilvo, 2023-10-13:
I will, thanks!

ruilvo, 2023-10-13:
Do you have an invite URL for that?

sizzlik, 2023-10-13:
> Do you have an invite URL for that?

https://discord.gg/hTkKXm52