portapack-mayhem/mayhem-firmware #1501: AMP is inverted: 0 means with 14 dB gain, 1 means 0 dB gain

alain00091, 2023-10-15:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
AMP is inverted

When AMP = 0   it should be    0 dB amplification in receive mode (no gain/amplification)
When AMP = 1  it should be  +14 dB amplification in receive mode

and now if it is 0   it is + 14dB ....

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

audio with good freq, see the RSSI bargraph ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

try it with    231013 bin or 1.7.4

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-10-15:
???

f1ghy, 2023-10-15:
https://github.com/eried/portapack-mayhem/wiki/preamplifier-ic-replacement#rx-problems

gullradriel, 2023-10-15:
Ho yeah. @f1ghy is right. You may have a fried amp somewhere.

alain00091, 2023-10-16:
YES you are right !!!,   it is a HARDWARE FAILURE/BUG
I checked with very old release of portapack, and also with SDRSHARP

What is amazing is **that it works** , I see the gain increasing if amp = 0 and decreasing if amp= 1
So the only problem is that it is INVERTED !!!
I will not desolder the RF Preamplifier u13 (MGA-81563) ,  it is too small for my eyes
Thanks a lot for your support

Brumi-2021, 2023-10-16:
Then @alain00091 ,  although you did not replace it , can we close that issue? 

Thanks

alain00091, 2023-10-16:
Yes you can close it and sorry