portapack-mayhem/mayhem-firmware #1508: Very slow SD over USB transfer speed

yellowz1234, 2023-10-17:
**Describe the bug**
Not sure whether this is a bug or a hardware/software limitation, but the data transfer speed when using SD over USB functionality is very slow.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Portapack to a computer and enable SD over USB function.
2. Transfer a large file to the SD card, e.g., world map.
3. The data transfer rate is typically around 177 KB/s (sometimes increases to 355 KB/s, but never higher).

**Expected behavior**
Because the HackRF has a USB2 connection, I would expect higher transfer speeds.

**Affected versions**
Observed on all firmware versions since the SD over USB function has been implemented.

**Additional**
Tested with:
- HackRF R9 and Portapack H2+R4.
- 128 GB and 512 GB class 10 microsd cards, formatted in FAT32 format, experimented with different cluster sizes.
- Tried different USB cables.
- Tried different computers and OS (Windows 11 x64, Debian 11 ARM64).
This did not improve the transfer speeds.

NotherNgineer, 2023-10-17:
It's not just you.  :-)  On Discord, other contributors have pointed out delay loops in the USB code (with could possibly be tweaked a bit).  The SD-over-USB code also doesn't work if baseband is compiled with link-time-optimizations, which could indicate a sensitivity to timing/code changes.  Personally I'm just happy it works!  :-)

yellowz1234, 2023-10-18:
Thanks for confirming that it's not just me.
Fair enough... I'm also glad that it works at all, albeit slow, so let's keep it that way.  
I'll close this issue.

NotherNgineer, 2023-10-21:
I would encourage you to keep this request open, but as an Enhancement request.

yellowz1234, 2023-10-21:
Reopening as request for enhancement instead of bug.

Brumi-2021, 2023-10-21:
@yellowz1234 , done as request , changed label to enhancement.

jLynx, 2024-01-15:
Is this still an issue @Brumi-2021  @yellowz1234 @NotherNgineer , or can we close this ticket now?

NotherNgineer, 2024-01-15:
With my fastest SD card and the latest nightly firmware and regardless of the "High Speed SD Card" setting, I see "SD over USB" speeds of ~434kb/s.  At this speed, it will take about 82 minutes to copy the world_map.bin file.  To me this still seems slow, even for USB 1.1, so I'm still hoping there's room for improvement.