portapack-mayhem/mayhem-firmware #1509: Battery voltage

DavidIgor, 2023-10-17:
Battery voltage indicator on top bar or even battery in percentile would be super cool to have ... and blue LEDs can be used for something else?! or just turned off.

gullradriel, 2023-10-17:
There is actually no way to have the indication of battery level from the firmware. You have to hack yourself something if it's really needed.
And we can not control the blue LEDs, you can only cut them. On recent models they are giving a hint of the battery level.

closed 2023-10-17 by gullradriel

DavidIgor, 2023-10-17:
yeah my ones are indicating battery level... but I wanted something more accurate

jLynx, 2023-10-17:
Salfy just not possible as nothing that can read the coltsge is physically connected to the HackRF. The blue LED's are controlled directly from the charge controller which is independent from the cpld on the PortaPack and also not connected to the HackRF