portapack-mayhem/mayhem-firmware #1510: ADS-B Bug?

execccuter, 2023-10-17:
Hello, 

actually i have an issue with the ads-b receive , it does not show any data, antenna is plugged coneccorrectly other signals are catching fine. 

ADS-B seems not to work... please check the feature or guide me what I could do wrong?

f1ghy, 2023-10-18:
Hi, I confirm, since 1.80 ADS-B doesn't work ;o/ (n_231016.bin woks fine, but not 1.80) this on my 2 PP. No mater of amplifier... Receive on waterfall, is ok (audio).

Herro95, 2023-10-18:
Yes it is a bug and it's not working. Please fix :-)

f1ghy, 2023-10-18:
n_231017.bin and n_231018.bin: ADS-B doesnt work fine
n_231016.bin woks fine

DavidIgor, 2023-10-18:
So it is bug ! here are my findings: 
if you close adsb app and go to transmit and transmit anything , then going back to adsb app and all works fine ! 

other things to note: 
-adsb app is not reacting to lna and vga settings only Amp is affecting receiving sigbal. 
-

yellowz1234, 2023-10-18:
Experiencing the same issue. May be related to bug #1468 .

Doing a transmit first does not solve the issue for me. Changing LNA, VGA or AMP settings makes no difference.

NotherNgineer, 2023-10-18:
I can reproduce the issue as well in n231017 and 1.8.0.

Comparing 10/16 and 10/17, the only interesting change was a 3-character longer name string for the POCSAG TX app, which shifted all the code after that in memory by 4 bytes.

I have to assume that the ADSB RX app, or its baseband code, is accessing some memory location that it shouldn't be.  This would also make sense that ADSB works after running some other apps, and could also explain #1468.

DavidIgor, 2023-10-18:
I have tested switching fw to latest hackrf and CPLD but no fix

NotherNgineer, 2023-10-18:
Oddly, if I build the latest code on my own system the ROM ends up being 2.5KB larger than n231017 and 1.8.0 despite there being no code changes, and it seems to work fine.  I don't know if the nightly builds might be using a different compiler version or other tools, TBD.  Example binary attached if anyone wants to try it.

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/13034419/portapack-h1_h2-mayhem.bin.zip)

Brumi-2021, 2023-10-18:
Hi, yes tried your binary , and at least I captured one plane . So it seems working . (but now not so many air traffic at this time) .

lucidliving69, 2023-10-18:
i tried that new binary and looks like it's working for me as well on the R9

yellowz1234, 2023-10-19:
> Oddly, if I build the latest code on my own system the ROM ends up being 2.5KB larger than n231017 and 1.8.0 despite there being no code changes, and it seems to work fine. I don't know if the [...]

@NotherNgineer This binary seems to solve the issue for me, as well as issue #1468. So glad you could finally reproduce it (by accidentally making it worse ;-) ).
Looking forward to an official fix now!

AC9DJ, 2023-10-19:
This fixed ADS-B for me as well. Hopefully the issue can get resolved in the next build. Thanks for digging into this further!

James P. Anderson

Sent with [Proton Mail](https://proton.me/) secure email.

DavidIgor, 2023-10-19:
same for TPMS app!!!!!!!!!

NotherNgineer, 2023-10-19:
Attached are the external apps from the same build as my test firmware above, so you don't have to make that *difficult* choice between ADS-B and Pacman.  You can have them both!   :-)

[APPS.zip](https://github.com/eried/portapack-mayhem/files/13039932/APPS.zip)

Still hoping someone can tell me which compiler version is being used for the nightly builds so I can figure out why the automated build failed and my build works OK -- with no code changes...

Herro95, 2023-10-19:
thank you now I can play pacman again and watch the planes :-) great!

f1ghy, 2023-10-19:
Thanks, we know that it is in the hands of a good Ngenieer ! "\_(~..Õ)_/"

DavidIgor, 2023-10-19:
apps folder has to be replaced ? I did but it did not worked for me . @NotherNgineer

DavidIgor, 2023-10-19:
For ADSB i tested previous fw and latest working is : mayhem_nightly_n_231016 
@NotherNgineer

NotherNgineer, 2023-10-20:
I've heard one positive report about that ADS-B RX might be working again in mayhem_nightly_n_231020.  We haven't actually done anything to fix the problem (the code changes are unrelated), so I'll keep investigating.

DavidIgor, 2023-10-20:
@NotherNgineer another confirmation from me ! it works !

f1ghy, 2023-10-20:
Hi,
The ADS-B of n_231020 woks fine here ! (in Marseille, France)
Thanks ;o)

yellowz1234, 2023-10-21:
I confirm that ADS-B reception still works well in nightly n_231021.

Additionally, the main problem as described in #1468 is solved as well.

The only scenario that is still not working for me is the following:
1. Start the Portapack and launch ADS-B RX (-> works)
2. Exit ADS-B RX and send a message through POCSAG TX (-> works)
3. Go back to ADS-B RX (-> ADS-B not working)

After exiting ADS-B RX and launching it again, it works... (reboot not required).

What I find strange is that the following scenario is working without problems:
1. Start the Portapack and send a message through POCSAG TX (-> works)
2. Exit POCSAG TX and launch ADS-B RX (-> works)

avltree9798, 2023-10-28:
Any update on when this issue will be fix in the latest release?

f1ghy, 2023-10-28:
> Any update on when this issue will be fix in the latest release? 

Hi, It is fix since the lastest nightly realase. the last is n_231026.bin compiled whith a new compilator updated, i think !
Probably, it should possible to close this issue.

NotherNgineer, 2023-10-29:
Closing because it's working currently, but the issue is not understood and will undoubtedly show up again in the future as code is moved around...

closed 2023-10-29 by NotherNgineer

LouDnl, 2023-11-16:
With the latest nightly this is still an issue 2023-11-15 and latest cpld from official hackrf repo this is still an issue.
I have to open ADS-B TX, press send and stop, go back to RX and then it works when I activate the LNA.
If I exit the app and enter it again it no longer works untill I press send in TX again.

NotherNgineer, 2023-11-16:
@LouDnl  Other users reported the ADS-B RX issue went away in nightly builds including n_231016, n231021, n231026 (see above).  Do you see the issue in those nightly builds that other users are having success using?  Possibly the issue has come back, or maybe there's a new issue in the latest nightly due to recent ADS-B RX code changes.

f1ghy, 2023-11-16:
Hi, I actualy still have this issue, or a same one, with an H1r9... where adsb detect nothing, wile my H2+r9 nicely  detect planes as well. Am I alone ?

LouDnl, 2023-11-17:
> @LouDnl  Other users reported the ADS-B RX issue went away in nightly builds including n_231016, n231021, n231026 (see above).  Do you see the issue in those nightly builds that other users are [...]

I will try those nightlies and let you know. I have this issue with the november 15th nightly.

zach67385, 2023-11-18:
Hey everyone, just received my HackRF portapack H2+ from opensourcesdrlab on aliexpress. everything working great however the ADS-b function isnt picking up anything, no matter what I do. Version 1.8.0. Is there no simple fix? dont want to accidently brick the device

LouDnl, 2023-11-19:
@NotherNgineer Could not find n231026 in the tagged nightlies so I tried n231021 and that worked directly without having to transmit first.
Reflashed with newest nightly n20231119 and it now works also.
It's worth noting that I need to enable the front amp for it to receive any signals.
```shell
Board ID Number: 2 (HackRF One)
Firmware Version: n_231119 (API:1.08)
```

amelillo, 2023-12-16:
It might be worth noting that I flashed mine with the latest nightly release (n_231216) and ADS-B rx works without problems.
`Board ID Number: 4 (HackRF One)`
`Firmware Version: n_231216 (API:1.08)`