portapack-mayhem/mayhem-firmware #1514: Mayhem & HackRF rev (R9) Transmit Bug Chinese version

Tanamichi, 2023-10-18:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases) (no nightly versions before issue)

----

**Describe the bug**
Transmitting not working properly, it emits only carrier no modulation.
Tested on all transmit modes, no effect.
Replay mode same not working.

On HackRF mode working with sdrangel work normally.

**To Reproduce**
Steps to reproduce the behavior:
1. Try transmit something, microphone, SSTV, etc...
2. See in another SDR receiver on transmitted carrier

**Expected behavior**
Working modulation and hear transmitted signal

**Affected versions**
* Latest Stable release: 1.8.0
* Latest Nightly tested release: n_231017
* Previous working release: i don't know.

**Additional**
I'm tried all solutions found here, but transmit not working as well.
Reference source changing not affect on transmit. I'm using Portapack TCXO.
[image]
Gain and AMP work properly, no burned parts ;) 
Also tested with full output power, still no effect.

Peripherals on screenshot: 
[image]

SD card test looks well(experimental 16gb)
[image]

any ideas to get working transmit mode on portapack?
Also found bug: receive not working after transmitting something, it's start working after reboot. if needed i will open a new issue.

DavidIgor, 2023-10-18:
@Tanamichi how one know if rf component/blown?

Brumi-2021, 2023-10-18:
Pls check #466 , and let us know if it solves your problem .

Cheers

viorellu, 2023-10-18:
I can confirm replay with Portapack not working. Whenever I record something from my walkie talkie and send the signal back only noise is heard.
Strangely enough, I can capture POCSAG messages from and replay them to a flipper zero consistently.
@Brumi-2021  Thank you , I will look into updating the SPI firmware.

Tanamichi, 2023-10-18:
> @Tanamichi how one know if rf component/blown?
try set different Gain or AMP values, and check these values affects on signal reception or transmitting
most blown IC in hackRF is amplifiler in RF part.

Thanks @Brumi-2021 i will check now.

Brumi-2021, 2023-10-18:
Thanks very much for your confirmation  ! 
Very happy and glad ! 👏👏👏👏
Then let's close it

closed 2023-10-18 by Brumi-2021

sipcz, 2024-02-08:
Hi, I have a similar problem with my device! Is it possible to fix it on windows? Thank you for your reply