portapack-mayhem/mayhem-firmware #1515: TPMS app picking up key fob

DavidIgor, 2023-10-18:
toms app is not working with my schrader sensors anymore (latest FW) instead app picking up key fob signals! Renault Dokker 2018. 
note: 
not picking my tpms sensors was present even in 1.7 versions... last time I have seen my schrader sensors data in tpms app was back with latest 1.6 FW!

DavidIgor, 2023-10-19:
also tpms app has same issue as in #1510

NotherNgineer, 2023-10-19:
I have tested TPMS with both 315Mhz and 434MHz sensors and I'm not seeing any issue on my H2R4+ running firmware version 1.8.0.

Also, I suspect that it's not related to #1510 since it didn't just start happening with the 10/17/23 nightly (and most likely won't be resolved by using the bootleg I've attached to #1510).

I would recommend going back through the past nightly firmware releases to determine exactly when it stopped working.

DavidIgor, 2023-10-19:
@NotherNgineer Ok I will try to find exact version it worked last ..... 
(already found that for my adsb app it was version mayhem_nightly_n_231016)

sizzlik, 2023-10-24:
Is TPMS standart in some cars or is that an extra only a few people install? Living in germany we got all sorts of cars from old tracktor to luxus limo, but i was never able to pick up anything while sitting next to a traffic light and check the cars on the red light. or while walking along a row of parked cars. (using the EU preconfig ~434mhz range)

DavidIgor, 2023-10-24:
@sizzlik TPMS system is used from like 2005-6 ish in almost every car. However TPMS that communicates via radio frequency is only one type of tpms system. nowadays tpms is using wheel speed sonsors to determine if tyre is flat or not. 
and you being in germany should definitely pick up tpms signals with tpms app! set it up to 434 (or 433 in old app). 
note: 
if you not getting anything it may be same issue I experiencing.

NotherNgineer, 2023-10-24:
@DavidIgor

What do you mean when you say it's picking up your keyfob?  Does the app show pressure and temperature for your keyfob (strange), or do you just see a bump on the signal strength indicator?  Many keyfobs use the same 315/433 frequencies as the TPMS sensors, so a bump on the signal strength indicator is normal.

Parked cars in a parking lot do not transmit TPMS signals.  TPMS sensors are activated by movement of the vehicle (still intermittently maybe every minute or so) or when tire pressure changes rapidly (e.g. when a blowout occurs).   But you should pick up some when you're driving near other cars or near traffic at a red light.

Also note that the TPMS sensors transmit data in various different formats, and not all are supported by the TPMS app.  There's an old enhancement request to add support for more sensors (RTL_433 supports some others that Mayhem doesn't).

If you're finding that the TPMS app isn't picking up any at all, you might delete the SETTINGS/rx_tpms.ini file so that any receiver settings will go back to the default value.  If the issue you're seeing showed up in a certain firmware version, please do go back and determine which firmware version.

Also FYI the "434" MHz displayed in 1.8.0 firmware is the same as "433" in earlier firmware.  In all cases it's referring to 433.92 MHz (I just rounded up the number on the display).

NotherNgineer, 2023-10-24:
If you have a TPMS sensor that is no longer installed in the vehicle, you can usually shake it to get it to transmit.  If it's in a vehicle, driving will make it transmit, or maybe letting air out of the tires.

NotherNgineer, 2023-10-24:
Note that more captured data will appear in the LOGS / TPMS.TXT file than appears on the screen (some captures are filtered out and discarded due to errors).  I would be interested in seeing the log data from a firmware version that works, and to compare that with the log data from a version that doesn't work.  It might also be interesting to see what goes into the log file when you activate that keyfob.

sizzlik, 2023-10-25:
Thanks to you both..now i learned a bit more about tpms. I got neither license nor a car, so i didnt know how they work or whats available out there. But as nothengineer mentioned RTL_433 i wonder if its much work to make a similar app like the tpms app to read those little weather stations for the garden? When i start RTL_443 with a simple rtl-sdr i pickup many of the surrounding neighbourhood devices and can read what their pool temp in the basement is etc. Would that be much work to implement, or does RTL_433 something special that makes it impossible?

NotherNgineer, 2023-10-25:
rtl_433 on PortaPack has been discussed in #113  (TLDR summary:  it won't be easy)

DavidIgor, 2023-10-26:
@sizzlik @NotherNgineer 1486 for me it is enough if tpms app will be duplicated and bit changed. so that it will display data from weather stations. don't need full blown rtl433.

sizzlik, 2023-10-27:
> @sizzlik @NotherNgineer 1486 for me it is enough if tpms app will be duplicated and bit changed. so that it will display data from weather stations. don't need full blown rtl433.

Yep, thats what i had in mind aswell. Not a full replica of rtl_433, just the existing tpms app modified to read some of the basic values like temp, humidity, barometer. But i know about those weather stations as much as i know about those tpms sensor..is the hard part to decode the signal or are there too many different protocols in use? You guys know best..would just be a nice addition to have, as many people here got some sort of weather sensors/stations/controls on the 433mhz range around their home.

tony1492, 2023-12-25:
I was wondering how many cars does it store? I was only able to hold only 66 readings and it would not hold anymore. So does it have a cap off point? Thanks

NotherNgineer, 2023-12-25:
> I was wondering how many cars does it store? I was only able to hold only 66 readings and it would not hold anymore. So does it have a cap off point? Thanks

Good question; thanks for testing this.  As far as I know, I think the app just keeps adding entries to the list until it runs out of memory and crashes...