portapack-mayhem/mayhem-firmware #1528: Radiosonde MeteoRadiy MRZ-N1

DiX880, 2023-10-24:
It is possible to organize the reception of weather satellites (MeteoRadiy MRZ-N1 (400 MHz)) ?
 For example , as in https://github.com/projecthorus/radiosonde_auto_rx/wiki
[image]

DiX880, 2026-10-27:
It hasn’t even been 3 years since I registered to ask a question here.
Special thanks to @Xmoo26 for the work done!

[image]

I think we can drop the subject.

Xmoo26, 2026-10-27:
@htotoo this one can also be closed.

@DiX880 Thank you for testing.

closed 2026-10-28 by htotoo