portapack-mayhem/mayhem-firmware #1529: Rolling Codes app

DavidIgor, 2023-10-25:
rolling codes app that recognizes protocols,can sinc with remote or set Hackrf PP as new remote (if access to sync button on garage door or gate ...). 
note: Im aware that we can not recognize everything, but at least get on Flipper Zero level would be fine. 

usecase: I would like to use my portapack as extra remote to my garage door,gate and doorbell that uses rolling code. Flipper is able to recognize protocols and be used as another remote by synchronizing with receivers.

gullradriel, 2023-10-31:
Use your flipper then.

closed 2023-10-31 by gullradriel

DavidIgor, 2023-10-31:
@gullradriel absolutely perfect answer! You are amazing dude... I was waiting for someone like you tell me that Im a moron and HRFPpH2+ has no processing power for something like that! 
But ehj ... why do we even use portapack? let's just stick with hack rd and usb cable connected to Pc! 
Have a good one 
@NotherNgineer

gullradriel, 2023-10-31:
There is actually no work at all done in that way.
Feel free to push us a Pull Request.
And if you read it correctly I closed it as 'not planned'.
And thanks for the praise.

gullradriel, 2023-10-31:
And my comment about the flipper was not sarcastic.
It's taking less space and already do what you want, or so it seems.

DavidIgor, 2023-10-31:
whatever

DavidIgor, 2023-10-31:
this answer of yours should fit earlier in the chain. :-) (sometimes it feels here like on reddit)

- I don't want roll-jam
- copy that 
- copy that

gullradriel, 2023-10-31:
You want rolling code sync.
It's not available in Mayhem until someone do the job, if ever it is possible.
Roger ?

mlody1983, 2026-10-24:
[army green.json](https://github.com/user-attachments/files/33903863/army.green.json)