portapack-mayhem/mayhem-firmware #1530: Some Question to the Update under Win

ju-funk, 2023-10-26:
**Describe the issue**
1. Under win the app "hackrf_update.exe" update the Device, i can not find the source to hackrf_update.c, 
    I see it not using the hackrf-lib, it using the win-lib  WinUsb.dll?!

2. When make a backup from the old-version under win: 
    1. Switch to hackrf-mode and 
    2. hackrf_spiflash -r old_firmware_file.bin
    is that the right way under win? 

**My Hardware**
hackrf_info.exe output:
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000419862xxxxxxxxxx
Board ID Number: 4 (HackRF One)
Firmware Version: v1.7.0 (API:1.07)
Part ID Number: 0xa000cb3c 0x006xxxxx
Hardware Revision: r9
Hardware does not appear to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

Can i update this with the 1.7.4? In the past it has trouble with "r9" but version 1.7.0 and higher should by solved, i think?
Have made a request to the seller for  the update but now after almost 3 months no answer received

3. Some little Bug with version 1.7.0
   1.switch on the device
   2. switch to Hackrf Mode
   3. start with hackrf_transfer -r data.cs8 -f 433920000 -a 0 -l 32 -g 32 -s 2000000
   4. only get a noise, no signals (press often a Remote-Control)
   5. end the transfer, an start again, then it works and get the signals
   6. with my own app, switch between receive, send and receive, that solve the problem too  
   7. in the app "Capture" no problem, works without workaround 

4. At first I was confused because in the app "Capture" the rate is displayed with 250k, but then internally multiplied by 8 which I discovered later, but why not displayed the right value?

Sorry for the lot of question, and thank you for the answer

ju-funk, 2023-11-17:
Okay, no answer, 
for point 2 I have the answer, i can make a backup from v1.7.0, and when compare with "portapack-h1_h2-mayhem.bin v1.7.0"
it is the same, in this case I can make the update.

But can give me a hint to point 1 were the source I can found for "hackrf_update.exe"

jLynx, 2024-01-15:
Hi, there is no source for the updater as it was developed close source  by one of the original developers. Are you still having issues?