portapack-mayhem/mayhem-firmware #1536: Recon App Cannot Read Input Files in Subfolder

tomzhu1024, 2023-10-30:
**Describe the bug**
When the input file is located in a sub-directory in the FREQMAN folder, its content cannot be read by the Recon app

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Recon app
2. Tap on 'CONFIG'
3. Tap on 'select input file'
4. Choose a file in a sub-directory in the FREQMAN folder (e.g. /FREQMAN/ABC/DEF.TXT)
5. Tap on 'SAVE'
6. The Recon app shows '...empty file...'
7. When I copy the file to the FREQMAN folder, selecting both the original file and the copied file works (looks like the code only takes the file name into consideration and assumes the path to be always /FREQMAN/)

**Expected behavior**
The Recon app should be able to read files from sub-directories in the FREQMAN folder

**Affected versions**
* Nightly Release - 2023-10-29

gullradriel, 2023-10-30:
Hello.
We can not read from subfolders as it would break the way it was designed.
Each file in FREQMAN directory is considered a category, as you can see per the GUI.
It's not a bug, it was designed that way.
I'm going to leave it as an enhancement request instead.
Not sure anyone is going to mess with this thought.

tomzhu1024, 2023-10-30:
I see it. Thanks for your explanation!

gullradriel, 2023-10-31:
Renamed the issue as it's not a Recon only related problem.

jLynx, 2024-01-15:
@gullradriel  can this be closed as its per design?

gullradriel, 2024-01-15:
Let's close it.
I will put up a clean enhancement request in the day.

closed 2024-01-15 by gullradriel