portapack-mayhem/mayhem-firmware #1539: SD card read speed

mattiarizzi, 2023-10-30:
Hello,
I would like to replay an IQ file having a sample rate of 8/10 MSps. With 8 bit encoding it is around 16/20 MB/s. 
I checked that LPC4320 can do it (52MHz*4bit= 26MB/s). But I dont know about the filesystem overhead. Is it possible to reach  more than 16MB/s?
Thank you!

Brumi-2021, 2023-10-30:
Hello @mattiarizzi ,
I have two different affordable brands  of micro SD cards V30 type ,

And with both , going to Debug —> SD card test , Checking last line I got 

Read speed : 
max around 10.1 or 10.5 MB/s , 
average between 6,2 to 6,5 MB/s 

I do not have any V60 or V90 ,maybe it is slightly better ???? , but looking my test results , I guess that replaying long files , the limit will be around the avg 6,2 to 6,5 MB/s.

if some user has better results , pls share them ,

mattiarizzi, 2023-10-31:
Thank you @Brumi-2021 !
6.5 MB/s is not enough :( 
I might need to go for raw reading. Maybe the limitation comes from the block size?

mattiarizzi, 2023-10-31:
Maybe it is a block size limitation:

ffconf.h content: 

#define _MIN_SS 512
#define _MAX_SS 512
/* These options configure the range of sector size to be supported. (512, 1024,
/  2048 or 4096) Always set both 512 for most systems, generic memory card and
/  harddisk. But a larger value may be required for on-board flash memory and some
/  type of optical media. When _MAX_SS is larger than _MIN_SS, FatFs is configured
/  to variable sector size and GET_SECTOR_SIZE command needs to be implemented to
/  the disk_ioctl() function. */

512 bytes is very small. Why? DMA limitation?

Brumi-2021, 2023-10-31:
Hi @mattiarizzi , I do not know .   Maybe other developers  knows it , because most likely we do not have here the original one developer.
I know that the code is having core signal buffers of 512 complex samples ,x 2 (I, Q) ,from ADC or  SD card, ..., for FFT waterfall, or filtering and decimating , .. , ... and storing or reading them .   But I do not know your direct question.

bernd-herzog, 2023-11-14:
i attached an oscilloscope to the CLK line of the sd card slot and it says it runs at 25MHz.

[image]

The code also confirms that we are running at 25MHz and the option for 50MHz is not enabled:

[image]

There might be a lot of headroom, but the TODO in the code indicates that it might not be as easy as only activating the fast mode.

bernd-herzog, 2023-11-14:
Ok, I tried that option. It works on my device. Here is the result:

Before:
[image]

After:
[image]