portapack-mayhem/mayhem-firmware #1544: Auido File Problem

neverbetter34, 2023-10-31:
Hello, I recently purchased the HackRF Portapack. It came with firmware v1.7.0 installed. Then I installed v1.8.0.  When I upload any audio file to the device, it says that the file must be .wav 16 bit. However, it said that a .wav 8-bit file should be uploaded.

Since I received this warning on the device, I tried many versions such as .wav 16 bit LE BE, sign unsigned. But I can't get any results. What should I do

sizzlik, 2023-11-01:
wav file for which app? Soundboard?

neverbetter34, 2023-11-01:
Yes soundboard. I want to play an audio file at any frequency but I get this error. The .wav file does not appear in Soundboard anyway. When I go into Wav viewer and click on the file, the error message I mentioned appears.

The device initially came with v1.7.0 version installed. There was this error in that version. Then I installed v1.8.0 and I still get the same error. I would be very happy if you can help

neverbetter34, 2023-11-01:
Has anyone else encountered this problem?

Brumi-2021, 2023-11-01:
Hi @neverbetter34 , did you try first with the sound board wav file samples that came in the copy-sd card folder ?

neverbetter34, 2023-11-01:
There was no sample audio file on the SD card. In addition audio files that I put in the wav folder in .wav format are deleted when I take out the SD card. I need to put it in the main directory so that it does not get deleted.

neverbetter34, 2023-11-01:
[image]

Brumi-2021, 2023-11-01:
That is why I told you , that you need to go some official release like 1.8.0 , download also mayhem_v1.8.0_copy_to_sdcard zip and unzip it and copy /wav contents to your SD card and you will have several compatible audio.wavs

neverbetter34, 2023-11-01:
I installed 1.8.0. However, .wav files are not copied. When I try to copy it myself later, I see that it disappears when I remove the USB reader and plug it in again. They don't disappear when I copy them to the home directory. But they still don't open in portapack

neverbetter34, 2023-11-01:
Could it be related to the SD card? SD card has 3.7 GB storage

Brumi-2021, 2023-11-01:
Hello , I think the easy way to copy it ,  is to remove the SD card from PP and put it to some usb adaptor and copy all those wav files to sd card /wav  from the laptop .

neverbetter34, 2023-11-01:
I tried your advice but it didn't work.  No matter how I copy the audio files to the SD card, the files are deleted when I remove the SD card.

[image]

Brumi-2021, 2023-11-01:
But i i think  that you are checking different path , not ? 
It should be /WAV , directly to the root not /SDCARD/WAV  , maybe that is the problem . 
You need to copy all unzipped contents to the SD Aroot with the same path folders.

sizzlik, 2023-11-02:
Hey brumi..how did you change the text color of the menu? Which source file do i have to edit?

neverbetter34, 2023-11-02:
> But i i think that you are checking different path , not ? It should be /WAV , directly to the root not /SDCARD/WAV , maybe that is the problem . You need to copy all unzipped contents to the SD Aroot with the same path folders.

Problem is solved. Thank you so much. There is only one problem left. How can I generate my own .wav file? I have Audocity. What should the format be?

gullradriel, 2023-11-02:
https://github.com/eried/portapack-mayhem/wiki/Soundboard
"Sounds should be 8-bit unsigned mono .wav files between 24000 and 48000 Hz"

closed 2023-11-02 by gullradriel

NotherNgineer, 2023-11-02:
> Hey brumi..how did you change the text color of the menu? Which source file do i have to edit?

The File Manager App selects color of each line based on the file extension.  This is hard-coded in the source code.

Brumi-2021, 2023-11-02:
Thanks @NotherNgineer 
@sizzlik , if your filemanager does not show those different hard coded  colours, probably  you are  still using 1.7.0 or older version   (from a previous commit)

neverbetter34, 2023-11-02:
> https://github.com/eried/portapack-mayhem/wiki/Soundboard "Sounds should be 8-bit unsigned mono .wav files between 24000 and 48000 Hz"

I do it as described in the link, but when I send the file to the device, I get a warning that it must be a 16-bit mono type file. Should I convert it to 16 bit?