portapack-mayhem/mayhem-firmware #1552: Add Scaling for MAX2837/2839 Temperature Sensor

NotherNgineer, 2023-11-05:
The temperature sensor values from the MAX2837/2839 IC's are not being interpreted correctly.  The output is not in degrees C, and needs to be scaled, per datasheet:

ON-CHIP TEMPERATURE SENSOR
Digital Output Code        Readout at DOUT pin through SPI
A4:A0 = 01011, D4:D0
T A = +25°C:  01111
T A = +85°C:  11101
T A = -40°C:   00001

Brumi-2021, 2023-11-05:
Hi @NotherNgineer , thanks for finding it .  I confirm you that it is currently a common issue in the hackrf r9 devices.
It should be just a fw incorrect  mapping .
I have checked both datasheets , and comparing the code , the code and process seems correct. 
But as  as the code said with comments, in max2837 datasheet it is perfectly specified the 5 ADC bits poistion in the register 7 (D4:D0) , but in the max2839 , it is confused.  So I believe that the 5 bits ADC are now in the max2839, in address 11, but maybe now ,  not D4:D0 as in the previous device .  I will also keep checking it .  

there is not as clear information as

NotherNgineer, 2023-11-05:
Funny thing is the scaling algorithm in ui_debug.cpp is wrong even for the MAX2837.  It should be this per the data sheets:
`temp_in_c = ((adc_sample_value - 15) * 4.4) + 25;`
But if I change the code to use this algorithm, it says the temperature of my H2R4+ is -5C, and says the R9 is 60C.  So then it's wrong on both boards...

closed 2023-11-06 by NotherNgineer

NotherNgineer, 2023-11-18:
This algorithm will need to be updated again after issue #1565 is resolved.