portapack-mayhem/mayhem-firmware #1553: BLE RX app discovers no devices (PortaPack H2+ with HackRF R9)

NotherNgineer, 2023-11-05:
The BLE RX app discovers no (or very few) devices when using the OpenSourceSDRLab PortaPack (PortaPack H2+ with HackRF r9).

NotherNgineer, 2023-11-06:
There seems to be a signal strength difference in the 2.4GHz band.  The one on the right has the HackRF r9 board.

[image]

f1ghy, 2023-11-07:
Thanks, it's same for me H2+, r9. Even H1, r9. even from transmiting packet from an portapacj, the other decode nothing. it receive (audio, glass, or level app) but decode juste few of it.

Tanamichi, 2023-11-08:
i have PortaPack H2+ with HackRF r9 and fully corfirm bug.
I also used 2.4G/5.8G antenna. No devices.
Checked on last night build n_231108

NotherNgineer, 2023-11-18:
Issue seems to be resolved by recent BLE RX app changes

Brumi-2021, 2023-11-18:
Hi @NotherNgineer , I agree with you that  we have solved  that original no reception problem in a r9 platform. But in my set,  I still see some lower sensitivity difference , and significant RSSI difference , specially in the red RSSI peak detection . How about yours ?

gullradriel, 2023-11-18:
In my case it is not that it is less sensible, or I can't measure it myself.
But RSSI values are unaffected by LNA and VGA. Only current db / power level is (green line in Level app) is affected.
That's what makes me thing it's a display problem or the way the new max is working/configured.

NotherNgineer, 2023-11-18:
> Hi @NotherNgineer , I agree with you that we have solved that original no reception problem in a r9 platform. But in my set, I still see some lower sensitivity difference , and significant RSSI difference , specially in the red RSSI peak detection . How about yours ?

@Brumi-2021  I'm seeing the same thing, and fewer BLE devices detected.  I'm reopening the issue.

NotherNgineer, 2023-12-09:
Closing this one I submitted regarding BLE (detected device count seems minor now), and opened a separate issue #1633 regarding the RSSI peak detection.

closed 2023-12-09 by NotherNgineer