portapack-mayhem/mayhem-firmware #1554: Flipper zero .sub files support? Also SsTV receive?

Nhscan, 2023-11-05:
Any chance of getting support of Flipper zero .sub files without needing to convert them on the PC?

Also a app to  receive sstv would be great. 

I have been converting.sub flipper zero files it works great I posted some on my GitHub but direct support would be great.

gullradriel, 2023-11-05:
First SSTV RX is an already opened issue, with a bounty: 
https://github.com/eried/portapack-mayhem/issues/255

Second, for the flipper .sub to .c16 format, nobody is working on it at the moment.
The firmware is in C++, in case you wanna try :-)

gullradriel, 2023-11-05:
I'm editing your title to keep the sub to c16 ref and remove SSTV from it. No offense, it's for issues clarity  :-)

amelillo, 2023-12-16:
> I have been converting.sub flipper zero files it works great I posted some on my GitHub but direct support would be great.

I took a peek at your repo but I'm wondering what do those c16 files actually do. I should tx them but which is their original purpose? I mean, what were they supposed to do in Flipper Zero and how should one use them in a HackRF/Portapack? Are they standard OOK gate signals? Cars? Whatever?

Nhscan, 2023-12-17:
I haven't gotten around to labeling each one but basically they're not all the brute Force files that are on the flipper zero trying to categorize everything and come up with a better way to organize. I tested if you and they work on our wide range of things depending on what files. I can't remember if I had the 315 MHz the 315 megahertz 24 but princeton works with a  brute forcing most of your cheap doorbells that you buy off of Amazon and GE wireless light switches . When I get time I'll start working on it again a little busy with the holidays and family hopefully I'll get them all converted and posted and listed in the right order now. My son was making a script that will Auto convert a batch of them at a time and hopefully retain the file name so there should be easy to identify.

NotherNgineer, 2024-01-29:
Perhaps it would be helpful to implement something like this .sub to .c16 converter in a Mayhem app: 
https://github.com/RocketGod-git/Flipper-zero-.sub-files-to-HackRF-Portapack-.c16-files-convert