portapack-mayhem/mayhem-firmware #1555: ADS-B mode not working. On latest version v1.8.0

RoDMain, 2023-11-05:
----

Hi Guys I have 2 x HackRF One SDR radios both with the Portapack H2 boards fitted, but I've noticed since upgrading both from v1.6.0 to the latest version v1.8.0 the ADS-B mode no longer works. I was wondering if this is a known fault with this version of Firmware. 
I suppose I could always revert back to v1.6.0 Firmware again and wait till another stable version comes out. try the latest nightly release before submitting this. 
You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

gsLukao, 2023-11-05:
This is true, after the update this function no longer works, I have already changed antennas, to different types and nothing works, is it just configuration?

Brumi-2021, 2023-11-05:
Hi  @RoDMain , you are completely right 
You can check , here , issue #1510 
https://github.com/eried/portapack-mayhem/issues/1510

If you use the latest nightly , it is running well,  ( and you will have some more additional  nice features) . 

Cheers,

closed 2023-11-06 by NotherNgineer