portapack-mayhem/mayhem-firmware #1562: Wiki points to GCC 9.3.1, but nightly builds/releases are using GCC 9.2.1

NotherNgineer, 2023-11-06:
There's an inconsistency in the compiler version being used for the nightly builds/releases, versus those used by firmware developers who followed the instructions in the wiki...  Would be good if they matched I think, for a developer testing perspective.

NotherNgineer, 2023-11-19:
FYI:  @jLynx

NotherNgineer, 2023-12-29:
I'm having second thoughts about this.  Developer testing with various GCC compiler and hardware versions demonstrates code stability, and helps suss out intermittent issues such as timing issues or uninitialized memory.

But, particularly if the nightly build system uses a different GCC version from most developers, a developer-testing phase may be helpful for future firmware release candidates before the worldwide release announcement.

I think adding the developer-testing phase would be more helpful to ensure quality than requiring all developers to use the same compiler version as the build system.

closed 2023-12-29 by NotherNgineer

jLynx, 2023-12-30:
Have you been keeping up with the latest discord conversation?