portapack-mayhem/mayhem-firmware #1565: DEBUG -> PERIPHERALS -> MAX283x register dump data shifted by 1 bit?

NotherNgineer, 2023-11-06:
All of the data displayed by the DEBUG -> PERIPHERALS app for the MAX283x chips seems to be shifted right by 1 bit, versus expected values in the data sheet.

Unknown if the MAX283x chip registers are programmed incorrectly, or if it's just an an issue with this debug app, or the data sheet is wrong.

Assigning myself to investigate.

gullradriel, 2023-11-16:
Any news on your investigations amigo ?

NotherNgineer, 2023-11-17:
Still looks like an I2C issue with Read operations, affecting both the MAX2837 and MAX2839.  My latest PR #1584 will be useful for testing as we attempt to fix it.

NotherNgineer, 2023-11-18:
I have found that slowing down the SPI timing for the SSP1 interface, by writing 4 to the SSP1 Prescaler address 0x400C5010 (using my debug code in PR #1588), temporarily resolves the issue until the next Receive operation.

I need to figure out where this register is being written; perhaps in the HackRF code.  Once this is fixed, we'll need to update the Temperature scaling algorithm as well.

gullradriel, 2023-11-18:
Nice things you found !
Thanks for your time :-)

Brumi-2021, 2023-11-18:
Nice investigation job @NotherNgineer !
But i could not follow you , what  is your current theory behind this ? That max2839 can not suppport the actual  max2837 SSPI communication speed ?  Or it is a common problem of them ?

NotherNgineer, 2023-11-18:
> Nice investigation job @NotherNgineer ! But i could not follow you , what is your current theory behind this ? That max2839 can not suppport the actual max2837 SSPI communication speed ? Or it is a common problem of them ?

It's a common problem to both chips.  Note in first comment that few registers are actually read from the MAX283x chips so the impact seems small.

PR #1590 (2023-11-18): Fixed SPI Read timing issue affecting MAX283x chips
Added SPI timing margin for MAX283x chips to fix #1565.

Note that HackRF code uses a clock divisor of 21 where Mayhem code was using a clock divisor of 4, so we were attempting SPI communications 5X faster than when the HackRF module is connected to a PC.  This change only increases the clock divisor to 5, so we're still 4X faster, but it seems to resolve the issue.

Also fixed Temperature scaling since the Temperature register is now being read correctly.

Test version available here on Discord:
https://discord.com/channels/7196697648044442 [... cut]
closed 2023-11-18 by NotherNgineer