portapack-mayhem/mayhem-firmware #1571: Frequency Detection with HackRF

neverbetter34, 2023-11-09:
Hello, I'm new to hackrf. I had a question like this:
Can we instantly detect frequencies with HackRF? For example, can I instantly detect the frequency of my phone signal while talking on the phone? Or do I need to scan the entire spectrum? I know tools like Search and Scanner, but they don't seem to work the way I want. However, although Flipper Zero is a much smaller device, it can do this. Can you explain the reason for this?

sizzlik, 2023-11-11:
You got to know what frequency you expect..everything on the spectrum got its place. The flipper detection is for ISM bands in the 430-440mhz range (differs a bit for US fobs) and only catches stuff in that range. You got to see in what range your target device is supposed to send signals and then you can sure pin point it. GSM band is around 900mhz to 1,8ghz, wifi is from 2,4 to 5ghz etc. You got to know what you look for. I guess "instand detection" of your phone signal is impossible as the spectrum is too wide with too much noise and too many variables for each country, protocol used etc. HackRF is a SDR, not a specialized tool. The flipper on the other hand has a small fixed spectrum to look at and it knows what to look for/interpret it. The basic frequency detection of a keyfob is also possible with the mayhem scanner app..set its bandwidth, set a threshold for the signal and press the button of your remote..the scanner app will log the frequency that broke the threshold..but phones is another story. Maybe invest a few bucks in a cheap spectrum analyzer like tiny/nanoSA and a near-field probe..then you just got to hold the probe next to your phone antenna and you can see the spikes on a nice graph.

gullradriel, 2023-11-11:
Use the Looking glass to watch for signals ranges, spectrum or level mode.
Use the Search or the Recon (recon can also autorecord during locking time, audio or raw)
There already is a list named RECON_F0. Guess what it is.

gullradriel, 2023-11-11:
Actually you have at least four apps to detect signals.
Read the wiki.
Ask discord.
Apps: Looking Class, Calls, Search, Recon.

closed 2023-11-11 by gullradriel

neverbetter34, 2023-11-11:
> You got to know what frequency you expect..everything on the spectrum got its place. The flipper detection is for ISM bands in the 430-440mhz range (differs a bit for US fobs) and only catches stuff in that range. You got to see in what range your target device is supposed to send signals and then you can sure pin point it with the waterfall graph. GSM band is around 900mhz to 1,8ghz, wifi is from 2,4 to 5ghz etc. You got to know what you look for. I guess "instand detection" of your phone signal is impossible as the spectrum is too wide with too much noise and too many variables for each country, protocol used etc. HackRF is a SDR, not a specialized tool. The flipper on the other hand has a small fixed spectrum to look at and it knows what to look for/interpret it. The basic frequency detection of a keyfob is also possible with the mayhem scanner app..set its bandwidth, set a threshold for the signal and press the button of your remote..the scanner app will log the frequency that broke the threshold..but phones is another story. Maybe invest a few bucks in a cheap spectrum analyzer like tiny/nanoSA and a near-field probe..then you just got to hold the probe next to your phone antenna and you can see the spikes on a nice graph.

Thank you for your detailed explanation..

neverbetter34, 2023-11-11:
> Use the Looking glass to watch for signals ranges, spectrum or level mode. Use the Search or the Recon (recon can also autorecord during locking time, audio or raw) There already is a list named RECON_F0. Guess what it is.

You can express what you want to say without speaking in imperative form

gullradriel, 2023-11-11:
People are sensitive these days.
I'm sorry I couldn't give you more informations in less words :-D

sizzlik, 2023-11-11:
Dont want to see gull flipping tables..people might get hurt (╯°□°)╯︵ ┻━┻

BuddyPage, 2024-01-12:
A real 6 ghz spectrum analyzer window, (in landscape mode?), in the Portapack, not HackRF mode, plus a true frequency counter would be nice too. The TinySA has slower sweep rate than the HackRF.