portapack-mayhem/mayhem-firmware #1576: Morse TX App keeps frozen when pressing START tx in r9 PP.

Brumi-2021, 2023-11-12:
Hi,  that problem was pointed by @bl33p  in discord  Mic tx-rx 2-way -App  channel. 

I confirmed that in my PP r9, using latest today "next" ,  12th of Nov,,   the Morse App TX freezes when pressing  START tx .  
But  In PP r6  H2+R4 platform , the transmission progression  bar is working well and not freeze at all . (we need to switch off - on to recover it ) 

I can confirm that using fw 1,7.4 same conditions, r9 also did not work there. (so probably it is an old problem with the PP r9 ) 

If anyone  has also PP r9 version working well  ,
pls leave here your feedback., because we do not know if it is a general r9 problem or related to some specific r9 model.
We need to investigate it . 

Cheers.

vanpit, 2023-11-15:
I can confirm this issue. H2+ / R9 freezes instantly when pressing Start in Morse APP when CW modulation is chosen. With FM modulation app works well. Tested with 1.8.0, last nigdtly builds and also with 1.7.0 - but here it is little bit different - device freezes sometimes after few clicks of progress bar and sometimes M0 Guru Meditation pops up.

NotherNgineer, 2023-11-19:
@Brumi-2021  Same issue here.  Freezes with TX LED illuminated on my H2+/R9.  No response to DFU button.

NotherNgineer, 2023-11-23:
The issue appears to be that the Morse app is trying to toggle "gpio_og_tx" (GPIO5_15) directly to modulate the carrier but this GPIO bit has a different meaning on HackRF r9 boards.

There must be a different way to modulate the transmit carrier on HackRF r9 (although I don't know what it is...)

Brumi-2021, 2023-11-25:
Based on my test in a r9 , I think that we can close that issue with above PR.

closed 2023-11-25 by Brumi-2021