portapack-mayhem/mayhem-firmware #1581: Jump to app button

gullradriel, 2023-11-16:
**Is your feature request related to a problem? Please describe.**

It's boring to jump from app to app. Each app has to make a lots of buttons to allow switch from Looking Glass to Capture i.e

**Describe the solution you'd like**

A configurable jump button in which we can see a list of available apps once clicked.

KenVanHoeylandt, 2024-02-02:
Perhaps via long press on the back button? That way you get it implemented automatically for all the apps.

An app could then define built-in shortcuts, but you could even allow users to add their own to the list.

htotoo, 2024-02-03:
I like the idea!
And we could add a settings app, to manage if someone want to open one kind always with a specific app.